bigtreetech/BIGTREETECH-TouchScreenFirmware #381: [FR] IDEA: report flash used

Andrei-Pozolotin, 2020-03-11:
since `spi flash` is currently a limited resource:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/364
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/365

then it would make sense to track its usage

code in prototype
[v-user-2/TFT/src/User/API/boot.c](https://github.com/random-builder/bigtree-tft-code/blob/v-user-2/TFT/src/User/API/boot.c)

example boot progress report:

* during font copy

[image]

* during icon copy

[image]

Andrei-Pozolotin, 2020-03-11:
also, post-boot report:

*code

[v-user-3/TFT/src/User/Menu/SystemDebug.c](https://github.com/random-builder/bigtree-tft-code/blob/v-user-3/TFT/src/User/Menu/SystemDebug.c#L64)

* report screen

[image]

thisiskeithb, 2020-03-12:
This would be handy for developers 🙂

closed 2020-08-17 by bigtreetech