bigtreetech/BIGTREETECH-TouchScreenFirmware #383: [BUG] (TFT35_V3 Extremely long update and booting time)

dineshvyas, 2020-03-11:
### Description

### Steps to reproduce

1. [First Step] update TFT35V3. it will take 10+ minutes to update after 99%
2. [Second Step] reset it takes 10+ minutes after every reset.
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

carl1961, 2020-03-12:
No issue with update on mine. 2 minutes max. I installed using sdcard.

guruathwal, 2020-03-12:
that does not look like a standard update screen. are you using any other modded firmware??

dineshvyas, 2020-03-15:
accidental uncommented debug msg.
removed and installed latest release.

closed 2020-03-15 by dineshvyas