bigtreetech/BIGTREETECH-TouchScreenFirmware #388: Zits when serial connected (SKR Mini E3 1.2 + BTT TFT Screen + Octoprint)

robstermn, 2020-03-13:
I am having this issue as well, with SKR Mini E3 1.2 + BTT TFT Screen V35 + Octoprint.
Printing from Octoprint, I get these small zits all over.
I've increased the baudrate from 115200 to 250000 with the same issue.

[image]

[image]

If I print from the TFT display via micro SD, prints fine???

[image]

[image]

guruathwal, 2020-03-13:
Duplicate of #282 
this was fixed in MarlinFirmware bugfix-2.0.x https://github.com/MarlinFirmware/Marlin/pull/17032
you need to update your Marlin firmware to the latest bugfix version.

JWolfJr, 2020-03-13:
I'm running the latest bug fix and still having the stuttering and blobs. I also have my octopi connected to the TFT from the gpio pins to uart3. Changed my baud rate to 250000, also enabled rapid communication. Is there anything I forgot?

RohanClegg, 2020-03-13:
Having the same issue. Please fix. Also, on the touch screen i don't have the ability to pause or adjust anything whilst octoprint is printing.

robstermn-64, 2020-03-13:
I'm using the latest Marlin from:
[https://github.com/morningreis/Marlin-SKR-E3-Mini-1.2](url)

And I've tried both versions with the same result pictured above.

Here are my configs for reference:

[https://drive.google.com/open?id=1d3piwB5Yty5qhABel1vJHKe91o0p0F_q](url)

guruathwal, 2020-03-14:
@robstermn-64 The Marlin firmware you are using is not the latest version.
 you need to get it from here: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

guruathwal, 2020-03-14:
Also, Marlin 2.0.5 was just released 2 hours ago. https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.5

CRCinAU, 2020-03-20:
Close the report... Use the right firmware and it'll go away...

JWolfJr, 2020-03-20:
I'm using the latest, it's definitely better. Only print so far still leaving blobs and zits is when I printed a lithophane. Both TFT modes. Otherwise no problems. If I ever decide to do more lithophanes, I'll just use the SD card. Thank you to all for the hard work.

bjoernwedi, 2020-03-21:
It is no marlin bug: had the same when printing over onboard printer sd.
Seems in marlin Mode the rx buffer is not read, so marlin get‘s Stück and flushes After a while, the continues. If you increase Marlins serial buffers it gets better, but at high speeds the issue Returns.
I would Renommees to read the marlin buffer frequently to ensure it does not exhaust.
This also explain, while there is no issue when printing from lcd sd, because them serial is used in Touch Mode, so buffer is read anyways.

bjoernwedi, 2020-03-21:
> Close the report... Use the right firmware and it'll go away...

On. It depends on your print Speed. When printing at 90mm/s or abviere, the issue returns

bjoernwedi, 2020-03-21:
For Proving the issue: If you use marlin Mode and print from onboard and (Printer Mainboard) while the Black serial tft cable is connected, you will See at Some speed the blobs starting.
Do not stop the print, just remove the Black serial cable and no stuttering/blobs will Appell anymore.

JWolfJr, 2020-03-22:
Nevermind, tried printing from a thumb drive and the TFT/printer locked up as soon as I tapped on the gcode file.  I know I've seen an open issue about this, but I can't find it.

jidayME, 2020-03-30:
SKR Mini E3 TFT 24, the same issue. If shown main screen I have zits. If I turn another menu than everything fine.

manuamador, 2020-07-12:
Same here, with previous Marlin and latest Marlin. Printing from onboard SD card is ok. With octoprint, zits and blobs.
Something is wrong with either the board firmware (from BTT) or the TFT firmware or both.

CRCinAU, 2020-07-12:
You haven't updated to the latest Marlin 2.x.x version.... Do that and the problem will be fixed.

manuamador, 2020-07-12:
I have the latest Marlin from this GitHub. My Pi is connected to the board. Should I Try to connect it to the screen ?
I may try without the tft screen. It sends commands that cause halts during the print.

Tomthi, 2020-07-22:
I'm seeing the same thing on a SKR 1.4 with a TFT24. When I print from Octo it see the most artifacts on curves, there is a slight hesitation on curves. The only way to get around this is to either disconnect the TFT in Touch screen mode, print off the SD card on the TFT or  print to Octo and switch to emulation mode. I've tried a bunch of stuff but this seem to be the only workaround.

closed 2020-08-19 by bigtreetech