bigtreetech/BIGTREETECH-TouchScreenFirmware #395: [FR] Preview models on Unified Menu

Bankinghelp, 2020-03-13:
Hi, would it be possible to add the preview model and view the model while printing with the unified menu? I would really appreciate that feature and will make my printer modern and nice! I know its in the classic menu but I really like the unified one! Thank you for considerating this!
Cheers!

guruathwal, 2020-03-14:
the feature is already in unified menu... you need to update to the latest firmware

Bankinghelp, 2020-03-14:
@guruathwal thanks I will check

Bankinghelp, 2020-03-14:
@guruathwal I seaid it because of this: "Gcode file under unified menu is in list mode, unable to preview the model, but more friendly for long filename"

guruathwal, 2020-03-14:
@Bankinghelp Unified menu has both Icon mode and List mode:
there is an option to enable icon mode in Feature Settings Menu.

Bankinghelp, 2020-03-14:
@guruathwal Ok it works, but I can't make the photo to the model. The tool is in Chinese and when I  load the model nothing happens just a Chinese message... what can I do?

Bankinghelp, 2020-03-16:
@guruathwal do u know how to use it, been 3 days trying with no success Help please.

guruathwal, 2020-03-18:
you can find the usage instructions here: https://github.com/bigtreetech/BiQU_Tool

Bankinghelp, 2020-03-18:
@guruathwal yeah I know but still the same issue, it's strange... And still, no way I can change de language...

Hukuma1, 2020-03-19:
Just take a photo of the model in your favorite slicing program (e.g. Cura) and place that image in same place on SD card where your model is.

Bankinghelp, 2020-03-19:
With the same name as the model? and which kind of extension (jpg, png, bmp...)?

Hukuma1, 2020-03-19:
For TFT35 it would be:
benchy_95.bmp (dimensions of file 95x95)
benchy.gcode

Hukuma1, 2020-03-20:
Did you get it work? If so please close the two issues. :)

dp250f, 2020-03-31:
It would be nice if there were a Cura plugin which would automatically generate these bmp preview files. The MKS one embeds png files into the G-Code, which doesn't work on BTT screens.

oldman4U, 2020-07-21:
@Bankinghelp 

Please close this ticket because this is available now. Thank you

closed 2020-08-19 by bigtreetech