bigtreetech/BIGTREETECH-TouchScreenFirmware #396: [Q] M150 ERROR ON TFT35 E3 V3.0

Bankinghelp, 2020-03-13:
i have that screen and all works good but when I want to change the RGB color of the screen says that there is nop M150 command found. Do I need to change anything in marlin? My board is SKR v1.4 Turbo. Thanks!

bigtreetech, 2020-03-14:
hello, please try "Settings->Freature->Rotary Knob LED"

Bankinghelp, 2020-03-14:
@bigtreetech yeah that works, I would really like it if I could add the feature that when is heating extruder change color and when it's finished the green color etc... Is that possible? Thanks

bigtreetech, 2020-03-17:
Yes, it's possible. Just move the implementation in Marlin to here.  I need to sort out the requirements, which will be implemented later, but not soon. It will have a lower priority

Bankinghelp, 2020-03-17:
@bigtreetech and what about model preview? I have created a post but nobody replies... I can't make it work. May u help me? Thanks
PS: Details are in the other post

closed 2020-03-18 by Bankinghelp

carl1961, 2020-03-18:
@bigtreetech  PRINTER_EVENT_LEDS code:

`

###################################################################
Configuration.h    Marlin
###################################################################
/**
 * RGB LED / LED Strip Control
 *
 * Enable support for an RGB LED connected to 5V digital pins, or
 * an RGB Strip connected to MOSFETs controlled by digital pins.
 *
 * Adds the M150 command to set the LED (or LED strip) color.
 * If pins are PWM capable (e.g., 4, 5, 6, 11) then a range of
 * luminance values can be set from 0 to 255.
 * For Neopixel LED an overall brightness parameter is also available.
 *
 * *** CAUTION ***
 *  LED Strips require a MOSFET Chip between PWM lines and LEDs,
 *  as the Arduino cannot handle the current the LEDs will require.
 *  Failure to follow this precaution can destroy your Arduino!
 *  NOTE: A separate 5V power supply is required! The Neopixel LED needs
 *  more current than the Arduino 5V linear regulator can produce.
 * *** CAUTION ***
 *
 * LED Type. Enable only one of the following two options.
 *
 */
#define RGB_LED
//#define RGBW_LED

#if EITHER(RGB_LED, RGBW_LED)
  #define RGB_LED_R_PIN PF9
  #define RGB_LED_G_PIN PF8
  #define RGB_LED_B_PIN PF7
  //#define RGB_LED_W_PIN -1
#endif

// Support for Adafruit Neopixel LED driver
//#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRBW // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     4       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 30       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  //#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif
[... 343 lines cut ...]
      #if HAS_LEDS_OFF_FLAG
        if (leds_off_after_print) {
          set_done();
          leds_off_after_print = false;
        }
      #endif
    }

  #endif // SDSUPPORT
};

extern PrinterEventLEDs printerEventLEDs;

################################################################################
Other related files  just search "PRINTER_EVENT_LEDS"

temperature.cpp      Marlin\src\module  
queue.cpp            Marlin\src\gcode
leds.h               Marlin\src\feature\leds\leds.cpp 

################################################################################

`

[image]

Bankinghelp, 2020-03-19:
#define RGB_LED_R_PIN PF9
#define RGB_LED_G_PIN PF8
#define RGB_LED_B_PIN PF7

@carl1961 Which pins are for the SKR 1.4 Turbo?