bigtreetech/BIGTREETECH-TouchScreenFirmware #397: Error updating firmware TFT35 V3.0

leanlibutti, 2020-03-13:
Hello,

I have a problem that the screen appears as follows after updating the firmware to version 0.25.2: 

[image]

I would appreciate if you help me to solve this.

Regards, Leandro.

Bankinghelp, 2020-03-14:
uuum check all cables ar connected properlly and in the right side. and in marlin u have defined ports 0 and -1.

leanlibutti, 2020-03-14:
with the previous version (0.25) that did not happen. So I think it's not Marlin's problem

leanlibutti, 2020-03-14:
also do not know why it appears in Chinese if I have it configured with English

bigtreetech, 2020-03-14:
hello, maybe you need also to update the fonts and icons in [here]( https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/TFT35)

leanlibutti, 2020-03-14:
that was the problem !! thank you very much for your help

regards

closed 2020-03-14 by leanlibutti