bigtreetech/BIGTREETECH-TouchScreenFirmware #404: Use the beeper for its intended purpose!

Aleks-MO, 2020-03-14:
**Use the beeper for its intended purpose!**
[image]
**Start of printing and end of printing I have a beeper on LSD . Can someone write it in the program?**
[image]

Msq001, 2020-03-16:
OK, Has been added to the my work plan and will be implemented tomorrow

Aleks-MO, 2020-03-16:
@Msq001  
Put the implementation in the program here.

Msq001, 2020-03-17:
hello, has be implemented in #415 , you can test it

Aleks-MO, 2020-03-17:
@Msq001   
Thank you very much, I will check.

closed 2020-08-19 by bigtreetech