bigtreetech/BIGTREETECH-TouchScreenFirmware #407: [Q] How to use model prviiew.

Bankinghelp, 2020-03-14:
first is in Chinese and there is no way to translate it and secondly, it gives me an error message when uploading gcode and nothing happens. Also, I don't know where to save it (where gcode is or SD card root...). Any help?

Bankinghelp, 2020-03-16:
Any help, please? Still not working

rozenrod, 2020-04-09:
You need to choose an STL model. Save the .bmp files (I save them to the desktop), then drop these .bmp files to the memory card (Do not use the biqu_tool generated by gcode)

oldman4U, 2020-07-21:
Hi.

Today there has been a new method released which allows to get the functionality without bmp directly from Cura export. Please try and close this ticket in case this solves your problem. Thank you

bigtreetech, 2020-08-19:
Hello, Please refer to [here](https://github.com/bigtreetech/Bigtree3DPlugin)

closed 2020-08-19 by bigtreetech