bigtreetech/BIGTREETECH-TouchScreenFirmware #412: [FR] TFT35 Iconset by Peter Mann and Acenotass - Unified Menu Material Theme

Acenotass, 2020-03-16:
I want to share with you a beautiful set of iconset created by Peter Mann. I took its set as a basis and created the missing icons, redrew the existing icons a bit.

[image]
[image]
[image]
[TFT35 Iconset by Peter Mann and Acenotass - Unified Menu Material Theme.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4340599/TFT35.Iconset.by.Peter.Mann.and.Acenotass.-.Unified.Menu.Material.Theme.zip)

sc44, 2020-03-17:
This icon set is really great. I've been using it since this afternoon and think it's great.
Thanks alot

Acenotass, 2020-03-18:
Thank you, I tried to make everyone enjoy my work.

Hukuma1, 2020-03-18:
Looks great! Should be listed as an option for people to know it's out there. Do we even have a section listing custom icon packs? Maybe we could add it to the front page readme that @thisiskeithb just recently awesomified?

jthomeo, 2020-05-31:
This icon set is great. Thanks for sharing. To be perfect, only 4 icons are missing.
e_1_percent, e_5_percent, e_10_percent and screen_info.
I've replaced global_nozzle for a copy of Nozzle and global_bed for a copy of Hotbed and it's better yet.

sc44, 2020-06-01:
Hello,
updated Version for the Vx.x.26 release. e_1_percent, e_5_percent, e_10_percent.bmp is included.

[TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4711905/TFT35.zip)

jthomeo, 2020-06-02:
That's a great addition. Tks!

bpivk, 2020-06-08:
I have a problem with the first three status icons. They show the default ones. I downloaded and Installed both packs separately and together and same issue.

sonny-c, 2020-06-17:
What firmware is recommended to flash with this icon pack?  I flashed BIGTREE_TFT35_V3.0_E3.26.1.bin and over 50+ icon failed to load.

jthomeo, 2020-06-17:
@sonny-c, I've corrected all file names in mine. Try this one.
[TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4789839/TFT35.zip)

sonny-c, 2020-06-17:
> @bpivk and @sonny-c, I've corrected all file names in mine. Try this one.
> [TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4789839/TFT35.zip)

Thank you!

DrumClock, 2020-07-25:
Hi @Acenotass 
do you have icone also for model TFT 5 V3.0?
There is another dimension of bmp.

Thanks

closed 2020-11-03 by stale[bot]