bigtreetech/BIGTREETECH-TouchScreenFirmware #416: [BUG]

Bankinghelp, 2020-03-17:
Hi (Copy from another post, I have the exact same issue)
I have an odd experience. I have the TFT35 E3 V3.0 with skr 1.4 Turbo. 
The issue is after I select a file to print I am taken to the screen to monitor the print but everything on the display side stops working let me put some color to what happens:
1. Display timer Starts counting. (that works)
2. Bed and Nozzle temps show 26/0 28/0 ( basically close to the room temp) The requested values do not get there and the actual never updates
3. The printer starts to heat up performs G28 and G29 accordingly and starts to print

If I try and cancel the print the screen freezes and displays " Loading" and remains there.

I have #define SDCARD_CONNECTION ONBOARD (set)
and #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Before attempt to print I can manipulate all settings and the display works fine. have not tested to print from USB or from LCD SD slot.

Everything works fine if I use the simulated graphical display.

bigtreetech, 2020-03-18:
hello, when you print from onBoard SD card，and when the Marlin execute “M109” or "M190",  it will not respond TFT until the temperature reaches the target, “M105” query temperature of TFT does not work now, so
> 2. Bed and Nozzle temps show 26/0 28/0 ( basically close to the room temp) The requested values do not get there and the actual never updates

and if you cancel print, the screen will back to gcode file list interface, it will send "M20" to get gcode file name from Marlin. in the same way, Marlin will not respond until the temperature reaches the target, so it's "Loading...".

After testing, if you set `#define SERIAL_PORT_2 0` in marlin, the temperature will be automatically feedback to TFT by UART0 when wait "M109/M190", and TFT can update the temperature in real time, but printing can only be cancelled after the "M109/M190"(temperature reaches the target).

This issue needs to add "M108" code in abort printing, and required "EMERGENCY_PARSER" in Marlin.

Now you can set `#define SERIAL_PORT_2 0` in marlin(So you can see the real-time temperature)， and cancel printing after temperature reaches target, this will be a normal operating experience.

And we will improve the problems related to onboard sd printing recently, very thanks for your feedback

Bankinghelp, 2020-03-18:
So I set #define SERIAL_PORT_2 to 0 and #define SERIAL_PORT  to -1? Also, just enabling EMERGENCY_PARSER will fix the loading until it reaches the temperature right? Thank you!

bigtreetech, 2020-03-18:
hello， if you set #define SERIAL_PORT_2 to 0 and #define SERIAL_PORT to -1, it can fix 

> 2. Bed and Nozzle temps show 26/0 28/0 ( basically close to the room temp) The requested values do not get there and the actual never updates

  the loading need both enable `EMERGENCY_PARSER` AND "M108" need to be sent to Marlin immediately when cancal printing, "M108" have not yet implemented in TFT firmware now.
  so you can't cancal printing before reaches the temperature, or it will freezes.

thisiskeithb, 2020-03-18:
Note: `EMERGENCY_PARSER` is not supported on `STM32` in Marlin at this time.

Bankinghelp, 2020-03-18:
Ok, thank you @bigtreetech and @thisiskeithb . I will try all of that and tell you. Looking forward for the M108 update! and thanks again for the fast reply!

closed 2020-03-18 by Bankinghelp

pepegragera, 2020-04-08:
> hello， if you set #define SERIAL_PORT_2 to 0 and #define SERIAL_PORT to -1, it can fix
> 
> > 1. Bed and Nozzle temps show 26/0 28/0 ( basically close to the room temp) The requested values do not get there and the actual never updates
> 
> the loading need both enable `EMERGENCY_PARSER` AND "M108" need to be sent to Marlin immediately when cancal printing, "M108" have not yet implemented in TFT firmware now.
> so you can't cancal printing before reaches the temperature, or it will freezes.

I have the same problem with the latest version of TFT35 V1.2 and marlin 2.0.5.3 when I launch a print from the LCD the TFT shows after a while "Error M105" does not stop printing but throws that annoying message, my configuration is:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0