bigtreetech/BIGTREETECH-TouchScreenFirmware #417: [BUG] TFT35 stops responding with red 'Busy processing please wait....'

reQu1em00, 2020-03-18:
### Description

Using an SKR mini E3 1.2 with the tft35 on an ender 3 pro. Random drop outs when I print from anything (SD card, usb, octoprint) and no mater where it is connected.   Also the TFT will freeze with the message 'Busy processing please wait....'.  It may accept one or two presses of an icon but then nothing.  This happens in moths modes.  

If I print from Ocotpring via the mainboard USB the print will complete but from any port on the TFT it will stop.  

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
it should continue to work!

**Actual behavior**
TFT freezes and will not perform any function without turning off and back on.

### Hardware Variant
TFT35 v2.0
Mainboard in Ender 3 is a BTT SKR e3 mini 1.2

### TFT Firmware Version & Main Board Firmware details
All firmware versions up to the latest (BIGTREE_TFT35_V3.0.25.2.bin)

### Additional Information

Manu512, 2020-07-21:
I've same issue.
With SK1.4 Turbo. In both mode (TFT or LCD) if gcode change fan's speed many time, the screen report Busy Processing please wait, on the screen fan's speed switch between speed and the board stay with a unkown speed until reset...

I'm testing without the RS232 cable and it's working correctly.

critter42, 2020-08-14:
I experience the same issue as the OP - same hardware (TFT35 connected to a mini E3 1.2.

oldman4U, 2020-09-03:
Hi

What is the status of the reported issue?

Please let us know or close the ticket in case you do not need it anymore

Thank you

Manu512, 2020-09-03:
Hello,

If I print without RS232 cable (in Marlin Mode) no problem. 

A bit of a shame especially that I took this screen to have a touch ...

oldman4U, 2020-09-04:
I hope I remember right that this has been related to the order in which the serial ports are defined in Configuration.h. Have you tried to change this?

mmaacckkii, 2020-09-12:
I have the same problem TFT35 V3.0 and SKR mini 2.0 all updated to last firmware , random Busy processing please wait... and printer stop print, touch screen work but no reaction to touched functions.

Lumute, 2020-09-20:
Same here with a SKR Mini E3 2.0 + TFT35 E3 v3.0 and a Raspberry Pi 4B connected using GPIO to the TFT35 UART3 port.

Wondering if its some problem with my Marlin settings / compilation or a bug with the TFT firmware...

Lumute, 2020-09-20:
@oldman4U this is what I have in my Configuration.h for Serial ports:

#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1

I just notice that the Configuration.h from SKR E3 Mini 2.0 repo has it the other way:

#define SERIAL_PORT -1
#define SERIAL_PORT_2 2

Is that what you are referring to? I'll swap them and see if that makes a difference...

mmaacckkii, 2020-09-20:
Thx i allready change this and testing ;)

oldman4U, 2020-09-20:
Hi.

Yes, I use 
#define SERIAL_PORT -1
#define SERIAL_PORT_2 2

... well I even use only 2, because I do not use the serial port on my SKR E3 DIP, but I read in another ticket that this helped some users so I thought it could eventually help.

;-)

mmaacckkii, 2020-10-11:
OMG , that works :) i change 2 weeks ago and working fine all time :) THANK
YOU :)

oldman4U, 2020-10-11:
You are very welcome and it is great to hear it helped. Maybe the other users with the same issue try it also and can print without this issue.

cmatthews, 2020-11-16:
@oldman4U - Sorry to jump in on this but could you elaborate at all on what the root cause of this bug is? I am trying to understand how the fix RE #defining SERIAL_PORT and SERIAL_PORT_2 works in practice. 

I am experimenting with RepRap firmware on an SKR1.4 and am getting the same "Busy..." error when I change fan speeds. The serial ports are defined differently in RFF (via config.g and board.txt) and am trying to figure out whether I can mirror your fix in RFF.

oldman4U, 2020-11-17:
Hi @cmatthews.

No reason to be sorry. Unfortunately I am only a user like you and not a developer, sharing what I have learned the last 1.5 years.

Serial Ports define which ports of the mainboard are used for serial communication and there are two ports available in Marlin. -1 seems to be the USB connector on most if not all mainboards and has to be activated for Octoprint and if you would like to print from your slicer or Pronterface. The other one is used for the connection to the TFT. Even they should be equal in terms of how they are used, for me it has been working best to setup the more important connection as the first connection and -1 as the second. I also use 115200 as baud rate and never had problems like described here.

With other firmware than Marlin I have no experience, so sorry, but I can not help further. ;-(

Kind regards

cmatthews, 2020-11-18:
@oldman4U - Thanks for taking time to respond. I will dig around in the configuration files for RRF to see whether there is any way that I can prioritise the serial ports.

oldman4U, 2020-11-19:
You are very welcome. I hope you can find the reason of the problem.

oldman4U, 2021-01-11:
@reQu1em00

With the last firmware updates, some buffers have been increased which could lead to improved stability. Because the user which opened this ticket does not respond any more, I will remove this issue from the Known Bugs list. In case you are a user with the same issue - using the latest version of the firmware - please start a new ticket. This ticket will be closed automatically soon.

Thank you

MarkyPocock, 2021-01-19:
Before you do that I am on the latest firmware for the tft and board and encountered it today while printing from the Skr mini e3 SD slot. The progress froze at 13% and that error appeared but the print did go on to finish hence the time so I let it do its thing.

So it is still a present issue...
[image]

closed 2021-03-28 by stale[bot]

Magiquall, 2021-04-13:
Hello,
I have the same problem, with my BTT TFT35, for the moment connected to an MKS SGEN L V2 while waiting for my SKR-2.

I reduced the speed from 250,000 to 115,200 it was less frequent, but still present.
I swapped the COM Ports and had no more worries, until I printed. A print of one hour thirty minutes and the printer is frozen 3 layers from the end.

When it is frozen, it is not easy to put it back into operation, you have to turn it off and then wait a little while before turning it on again, often twice in a row.

Whenever it was frozen there was "Busy processing, please wait" at the top and in red, but nothing at all, the screen remains blocked even after several minutes and as during the print it continues to heat up I have a hole in my print.

ScottyMakesStuff, 2022-01-06:
I've got three ender 3 Pro RRF builds with a Mellow Fly RFF E3 Pro mainboard and  I'm having problems with the BIGTREETECH TFT35 E3 V3.0 Touch Screens which all have been giving me "Busy processing, please wait..." messages after being on for a short time. I've tried different baud rates and S numbers 0 to 3 and found that 'M575 P1 S2 B57600' is the only line in my config.h file that works. Has anyone here had this issue or have an idea what I can do to fix it?