bigtreetech/BIGTREETECH-TouchScreenFirmware #418: [FR] Custom Icon Pack Helper?

Hukuma1, 2020-03-18:
Sorry I don't know how to do PRs. D:

Wanted to update the photo examples of the icon UIs here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

I made the assets in native resolution for my TFT35 (480x320, thanks @guruathwal!) so these images  are 200% upscaled. I only have the main menu screen, but if there's interest maybe I'll work up the whole menu topology for icon pack making ease.

**Classic UI**
[image]

**Unified Modern UI**
[image]

**Peter Mann/Acenotass Icons**
[image]

Btw I could not locate these icons, @thisiskeithb: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/374#issuecomment-597282516 Do you have a link?

thisiskeithb, 2020-03-19:
They were in a Facebook post a while ago and aren’t compatible with the latest classic UI since icons need to be created.

Keeping multiple icon packs up to date here (assuming they can be included under GPL V3) and working for each UI will be the biggest challenge if that’s the direction you want to head in.

Hukuma1, 2020-03-19:
Ah I see. No, I was just thinking of linking to a few that were maintained. Seemed like there was a bunch scattered all around. This way people have a choice and the creators get the most exposure by being seen on the official page.

thisiskeithb, 2020-03-19:
Congratulations on your first contribution 🙂 https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/421

I've updated the readme to use the above previews for Classic & Unified UIs.

Hukuma1, 2020-03-19:
Thank you sir! :)

Btw @guruathwal as I was making this UI grid from scratch I noticed the Unified graphic of InfoBox.bmp was 1 pixel longer possibly? Or did I goof my math somewhere? It's currently 215x95 for TFT35. Should be maybe 214?

[image]

guruathwal, 2020-03-20:
@Hukuma1 
The correct gap between the icons is calculated as:
Icon spacing = [ Screen_Width - (Icon_width*4) ] / 4
Icon spacing = [480-(95*4)]/4
Icon spacing = 25

Left & Right spacing will be = Icon Spacing/2 = 12.5
But there is no half pixel. So The LCD will place the first icon at 12px. Due to this, all the icons are shifted to the left side by 1 pixel.

Hukuma1, 2020-03-20:
Guess I should have asked before making it! Lesson I'll never learn. ;) Is the top and bottom spacing also 12? Thanks for all the info.

closed 2020-08-19 by bigtreetech