bigtreetech/BIGTREETECH-TouchScreenFirmware #423: [Q] (Your Question?)bigtreetech tft35 v2.0

Hxspider, 2020-03-20:
**hi.
 I am trying to update my bigtreetech tft35 v2.0 to version .25 an it succeeds,  kind of...it loads and apparently updates but the icons aren't showing correctly. 
They are only blank.
I tried the .24 update as well and then I got one Icon, the print icon showed correctly. 
What is wrong?*
A clear and concise question or detail you would 

**bigtreetech tft35 v2.0**

**
[image]
**

thisiskeithb, 2020-03-20:
Did you follow the [Update TFT Firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md#update-tft-firmware) section in the readme?

Hxspider, 2020-03-20:
Oh…never mind..works fine now…

Hxspider, 2020-03-20:
Except the print screen icons are distorted..

Sendt fra min Samsung Galaxy-smarttelefon.

-------- Opprinnelig melding --------

Dato: 20.03.2020 02:06 (GMT+01:00)

Ko: Hxspider <fjellvang@hotmail.com>, Author <author@noreply.github.com>

Kyosuke33, 2020-03-20:
I have a similar problem, i try to upload new. 25.2 firmware on my TFT35 V3, it load great but after that i have weird icône. With the .24 i have the problem only on the surveillance screen, the menu is good.
[image]
[image]

Hxspider, 2020-03-22:
Anyone? 
My print icons are still distorted...

[image]

Kyosuke33, 2020-03-22:
Same problem for me, i have to use Acetonass icon to make it work. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/428

Clowntrigger, 2020-03-27:
See my response to [Issue 430](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/430#issuecomment-604664793).

oldman4U, 2020-09-03:
@Hxspider 

Please close this unused old ticket

Thank you

closed 2020-09-05 by Hxspider