bigtreetech/BIGTREETECH-TouchScreenFirmware #428: [FR] (feature request title)

Acenotass, 2020-03-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Kyosuke33, 2020-03-20:
Look great. Need this to my TFT35 V3.0 ! Not compatible ?

Acenotass, 2020-03-20:
The set is suitable for all versions of screens 3.5 "

Kyosuke33, 2020-03-20:
Cool ! What firmware i need to use, the classic or the Unified ?

joymonkey, 2020-03-20:
I just added these to my TFT35 V2.0 
Looks great, thank you for your hard work!

Acenotass, 2020-03-20:
> Cool ! What firmware i need to use, the classic or the Unified ?
> I just use your bmp folder instead of the bmp who is in th TFT35 folder. And i keep the original font folder ?
Suitable for both: Сlassic or the Unified
Fonts are not required to be rewritten if you have already flashed the Ubuntu font, which I did for all screens.

Acenotass, 2020-03-20:
> I just added these to my TFT35 V2.0
> Looks great, thank you for your hard work!

Thanks, I tried to make everyone enjoy it!

Kyosuke33, 2020-03-20:
So, i just put the firmware and your bmp folder on my sd card ?

Acenotass, 2020-03-20:
> So, i just put the firmware and your bmp folder on my sd card ?
Yes, unzip just place the TFT35 folder in the SD card and reload the printer.
But I still recommend flashing a completely new firmware and fonts, the instructions are on the main page of the project.

Kyosuke33, 2020-03-20:
I try and it work great thank you !

Kyosuke33, 2020-03-22:
I see you change the title. You change/update something since 2 days i have dowload it or it's just the title ?

Acenotass, 2020-03-22:
I redrawn a few icons to better suit the overall style.

[image]

Kyosuke33, 2020-03-22:
Ok nice thank you.

ssombra, 2020-03-22:
> I redrawn a few icons to better suit the overall style.
> 
> [image]

Looks very nice, I'll try it, where can I download it? is the link at the beginning of the post updated with the latest icons? One question, would it be easy to add a Preheat TPU icon?

Acenotass, 2020-03-23:
Yes, the link at the bottom of the issue is already updated. The TPU icon is easy to draw, if TPU preheat appears in the firmware, then I will draw this.

Interplain, 2020-03-29:
An Excellent job with this, was going to look at doing it myself but. not have the time. Only a slight criticism is the two Icons that have an element of fun and that is the Babysteps and the Print Button. would have liked to see something with out the two feet. and the Beachy. Sorry to offend. something a bit more Engineering maybe.

Acenotass, 2020-04-05:
obsole

closed 2020-04-05 by Acenotass

HappyCodingRobot, 2020-04-05:
obsolete..? but why?

Hacksumile, 2020-07-09:
There's a lot of bmp icons missing in your tft 35 link for the tft 35 v3 around 36 + bmp files. I'm not sure if that was intentional but I thought I'd let you know. Thank you for your awesome contribution though!