bigtreetech/BIGTREETECH-TouchScreenFirmware #430: Bug screen update 25.2

Kyosuke33, 2020-03-20:
Hello, i receive my TFT35 v3 yesterday, and when i try to update it to the 25.2 firmware (i put 25.2 firmware and Tft folder with bump and fonts on my sdcard), the update work great but my screen icône are weird. When i try with the 24.2 firmware, the same thing happen but only on the surveillance screen, the menu screen work great. Thank you. 
[image]
[image]
[image]

Kyosuke33, 2020-03-20:
MAJ, it work with the pack of icon by Acenotass, but not with the classic icon.

guruathwal, 2020-03-20:
@Kyosuke33 you need to update both fonts and icons from unified theme folder

Kyosuke33, 2020-03-20:
I try the Classic Firmware 25.2 with the TFT35 Folder (bump/fonts) and I try the Unified Firmware 25.2 with the TFT35 Folder (bump/fonts) but both won't work.
I try both Classic and Unified firmware with the TFT35 Folder (only fonts) and the bumps folder from Acenotass and it work.
Weird^^

guruathwal, 2020-03-20:
you need to copy the full TFT35 folder to your SD card. after copying it should look like this:
if your SD card is on F drive:
the BMP folder F:\TFT35\bmp
the font folder F:\TFT35\font
make sure you reformat your sdcard to fat32

Kyosuke33, 2020-03-20:
My sd card is fat32 and i do that for the folder, but won't work until i change the bump folder by the one of Acetonass.

Kyosuke33, 2020-03-20:
The sd card
[image]
Inside the tft35 folder
[image]

dziabi3, 2020-03-20:
I have the same problem but when I use the bin file the problem disappears.

jmdog98, 2020-03-21:
bump because I have the same exact problem

guruathwal, 2020-03-21:
@Kyosuke33 Did you compiled the firmware or used the precompiled bin file?

jmdog98, 2020-03-21:
@guruathwal I'm on the precompiled bin. using Acetonass icons also as a fix for now

guruathwal, 2020-03-21:
some time the Firmware does not update the BMP files on reset. This is because of bad SD card formatting or a currupted file on SDcard. Try full formatting the card to both FAT and FAT32. and try resetting the screen again a few times if it does not update the icons.

jmdog98, 2020-03-21:
I'm currently in a print that is finishing soon but after I'll try what you suggested, but when updating I can still see the icons being flashed over but it looks as if its trying to force the unified menu. by reset do you mean pressing the reset directly on the screen or power cycling the whole printer?

guruathwal, 2020-03-21:
you can see the original unified menu icon being updated or not?
What do mean by:
> its trying to force the unified menu.

Are you not using the unified menu BIN file?
yes, by reset I mean just pressing the reset button on the screen.

jmdog98, 2020-03-21:
What I ment was that when I updated it using the classic files it looks like a messed up version of the unified theme (because it shows the xyz coordinates down the center like the unified theme does) not the classic almost as if it was thinking it was suppose to load the unified theme.

guruathwal, 2020-03-21:
@jmdog98 It is not messed up. BigTreeTech added the status screen to the classic menu also. it was in the recent update that they removed the status screen from the classic menu. The precompiled BIN files are not yet updated with the change.
for the current precompiled BIN files for the classic menu, you need to use the unified menu icons also.

jmdog98, 2020-03-21:
I just reread your first comment where you stated to do that and realized I misread it the first time and I'm assuming that is probably what it is. My print has 5 minutes left so Ill be doing that next. I appreciate the help

Kyosuke33, 2020-03-21:
I'm not a spécialist, don't know what kind i used xD I just put on the SD card the files i show you below. Try Classic and Unified, on two différents SD Card, but not working until i change the bmp folder by the Acetonass folder.
[image]

jmdog98, 2020-03-21:
@Kyosuke33 You have to use the bmp files that are for the unified theme when trying to use classic because of a layout change with the newest update so instead of using Acetonass icons just copy in the icons from the unified theme files

Kyosuke33, 2020-03-21:
Ok, i only try Classic firmware with classic bmp and Unified firmware with unified bmp. I did not try to mix both. I try that later, for the moment i keep the Acetonass because i like the screen xD

Clowntrigger, 2020-03-26:
I ran into the same issue with the Unified Menu icons.

Looking at [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/readme/English/How%20to%20custom%20icon%202019-11-11.pdf] there is a hard coded list of icons. I was able to rename onboad_sd.bmp to OnBoardSD.bmp and the icon would show up. I pulled the repository down and the boot.c file does not have the hardcoded list, it scans the directory for a list found in icon_list.inc. The file icon_list.inc has the correct bmp listed as ONBOARD_SD. Compiling this firmware and installing with the TFT35 directory from the repository restored my icons. 

So, there is a disconnect with the compiled firmware at [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT35_V3.0.25.2.bin] and the expected filenames in the bmp directory.

reQu1em00, 2020-03-27:
> I ran into the same issue with the Unified Menu icons.
> Looking at How to custom icon 2019-11-11.pdf there is a hard coded list of icons. I was able to rename onboad_sd.bmp to OnBoardSD.bmp and the icon would show up. I pulled the repository down and the boot.c file does not have the hardcoded list, it scans the directory for a list found in icon_list.inc. The file icon_list.inc has the correct bmp listed as ONBOARD_SD. Compiling this firmware and installing with the TFT35 directory from the repository restored my icons.
> So, there is a disconnect with the compiled firmware at Copy to SD Card root directory to update - Unified Menu Material theme and the expected filenames in the bmp directory.
> edit: sorry, i have no idea how to insert correct links/urls. *Figured it out.

This worked for me.  The BMP have been updated but a new version of the firmware has not been uploaded.  If you compile the firmware yourself from the source files it creates BIGTREE_TFT35_V3.0.25.3.bin which then works.

cyberaxcess, 2020-03-31:
Compiling and using the BIGTREE_TFT35_V3.0.25.3.bin fixed this problem for me.

oldman4U, 2020-09-03:
Hi.

Please close this old unused ticket

Thank you

oldman4U, 2020-09-05:
@Kyosuke33 

Close this ticket

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket

oldman4U, 2020-10-06:
Please help the community and close the ticket once you do not need it anymore. Thank you

Kyosuke33, 2020-10-07:
Sorry i forget

closed 2020-10-07 by Kyosuke33