bigtreetech/BIGTREETECH-TouchScreenFirmware #431: Add TPU preheat

Wallaal06, 2020-03-21:
I have added a TPU preheat button to the preheat menu. Looking at all the other preheat buttons, I have everything set up right but when I click on it, it does nothing.

carl1961, 2020-03-21:
Wallaal06 I'm trying to learn this too, I think there or other files also that tie too.  Like PreheatMenu.o   How do you edit that?

Wallaal06, 2020-03-21:
I don't know... I can't even get the file to open right.

carl1961, 2020-03-21:
I found out the PreheatMenu.o or files that where compiled by platformio, so there not from the source files.

Wallaal06, 2020-03-27:
Can i get some help on adding another preheat button to the preheat menu. I copied all the settings i could find from the ABS button but when I click on the TPU button it does nothing.

bigtreetech, 2020-07-04:
The Preheat menu can be configured by config.ini  file, and added TPU button in Preheat menu. so close here,

closed 2020-07-04 by bigtreetech