bigtreetech/BIGTREETECH-TouchScreenFirmware #434: TFT35 V3.0: counter values shown instead of position values after home

blupeace, 2020-03-21:
The main page shows counter values instead of position values after home command.
Steps to reproduce the issue:
* build BIGTREE_TFT35_V3.0 using VSCode and PlatformIO
* copy .pio/build/BIGTREE_TFT35_V3.0/BIGTREE_TFT35_V3.0.25.2.bin to SDCard
* copy ./Copy to SD Card root directory to update -  Unified Menu Material theme/TFT35 folder to SDCard
* insert the SDCard into TFT35 and flash the new version
* go to Menu->Movement->Home->Home
* wait the homing
* got to Menu->Movement->Move and perform any movement

Expected behaviour:
[image]

Actual behaviour:
[image]

The first occurence of this issue appears at commit a0e7296196bfdaf7530edcae01e7008fbdccb836

Powerprobot, 2020-03-21:
I can confirm the issue. I make a programming- fork for me and my version work but it is not for public. i wil work for a better recognition in the code but i am a novice programmer (I start with the commodore64 and coded hard with assembler but the time overrun me) I hope the original programmer will fix this issue soon. 
The reason is that Marlin would change the output from single Axis coordinates to coordinates with additional steps.

Powerprobot, 2020-03-23:
@blupeace you find my repositories here: [TFT35_1.2](https://github.com/Powerprobot/BIGTREETECH-TouchScreenFirmware)
Attention: the config is cofigured as TFT35 1.2 with some additions and changes for me.

blupeace, 2020-03-23:
Thank you very much. I am going to check it.
Unluckily I cannot try if it works right now because yesterday I burn my main board (I already ordered a brand new one).
Whenever I will receive the board, I will let you know if it works properly.

Nascire, 2020-03-27:
@Powerprobot :
I had the same problem as @blupeace - just implemented your changes on latest master and I again have the correct position shown - thanks for sharing!
Maybe create a PR as long as no other/finer solution is done?

closed 2020-03-27 by bigtreetech

guruathwal, 2020-03-28:
This issue was caused by recent changes in M114 reporting in Marlin firmware. See Marlin issue thread here: https://github.com/MarlinFirmware/Marlin/issues/17255

Powerprobot, 2020-03-29:
I see. I updated both, Marlin and TFT35. I will delete my fork, because it is not longer needed. The Updates work well.

PR #464 (2020-03-27): Add more printer setup parameter to Parameter Settings Menu, add feedback sounds, add error display during icon update & other bug fixes
- Add Max Feed rate, Max Acceleration, acceleration, Probe offset, TMC bump sensitivity to Parameter Settings.
- Add Buzzer feedback sounds for Error, Echo, Success. etc.
- Increase ACK buffer size to 2048 to allow parsing of M503 reply.
- BugFix Baudrate.bmp not updating.
- Add displaying of BMP icon update success count and fail count.
- Add display of update error (BMP icon name & cause of the error).
- translate some comments into English.
- BugFix displa [... cut]