bigtreetech/BIGTREETECH-TouchScreenFirmware #444: TFT24 printing from usb drive does not work

asfandali, 2020-03-24:
TFT24 touch does not print from usb drive, usb does show up inside lcd but after selecting gcode nothing happens.  
printing from sdcard in classic marling mode works. 

i am using the latest lcd firmware.

j7777, 2020-03-27:
I confirm the same problem, I tried different flash drives.

stefi01, 2020-03-30:
same for me

bogus105, 2020-04-01:
not working for me as well. In general - serial/usb/SD things are not described well in the documentation so correct setting may be a challange. My setup consists of 2.4" bigtree TFT controller with build in USB stick reader as well as SD card reader. None of then is working for printing. Updating TFT controller firmware is ok. 
Same with SKR 1.3 - i can print over USB cable with PC as a host but i can;t manage to print without PC - using onboard SD card reader...

pljeroen, 2020-04-04:
Same issue. Previous version didnt have this problem, downloaded a month ago or so. Updated yesterday to newest version, cant print from USB stick connected to screen.

Havent tested SD or microSD, as I dont have any laying around atm. Printing with octoprint or pronterface works fine, connected through USB.

Sanguium, 2020-04-05:
Same problem here, TFT24, SKR1.3.
Printing from tft SD works fine, trying to print from USB it get stuck in 'Loading' after selecting the file.
Reverting to tag Vx.x.25 works.

Sanguium, 2020-04-05:
Did some testing, this is the commit that broke it, previous one works fine.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/81fb18a432347c2f0e84f3179c9f41004c1e2ddd

Sanguium, 2020-04-24:
Just tested latest version 62dbad, it works again for me.

closed 2020-08-19 by bigtreetech