bigtreetech/BIGTREETECH-TouchScreenFirmware #447: [BUG] No connection to the motherboard with latest updates

GenaUser, 2020-03-24:
Since the March 13 update, my TFT35 V3 can't connect to the board with the firmware compiled by me (« No printer attached » message).
The baud Rate is correct (115200 on board and TFT).
I compiled on Windows and Mac OS without making changes in the firmware for the same result.
All releases older to March 13 works properly.
Strangely, precompiled versions available on GitHub works.
Is there a bug, or did i miss something?

stefi01, 2020-03-30:
i get the same thing, touch was working untill i upgraded the firmware, now it says no printer attached, works fine in in the simulator mode

GenaUser, 2020-03-30:
I don't know if lots of people have the same problem. 
The 0.25.3 update on the TFT35V3 can't connect to my motherboard (Rumba board). 
The 0.25.2 firmware worked perfectly until 03/14/20 update.

stefi01, 2020-03-30:
i have the tft35v3 and the tft24 v1.1, they were both working fine until the latest update and now they both say no printer attached, they both still work when i switch to the 12864 so im using that for now

AntoszHUN, 2020-03-31:
He won't join me either. TFT: TFT35 V3.0 Motherboard: BTT SKR V1.4 Turbo was good with previous updates. It's not good now, neither is it good for you. :(

stefi01, 2020-03-31:
this will all be ignored too as bigtreetech ignore more or less all concerns with many bad reviews on how bad their customer support is, if we buy things then we expect them to work!!!!

AntoszHUN, 2020-04-01:
It worked so far. There was something wrong with the new update only. But I can't figure out what.

AntoszHUN, 2020-04-01:
There were many at the 250000 baud rate. Only 115200 works.

AntoszHUN, 2020-04-01:
I just downloaded the firmware from the site. Fixed a bug in the display. I put it on the screen. All baud rates are set at 115200. The display indicates that no printer is connected. But I can move the motors. But I don't see the temperatures 0/0. Is there a solution?

jimmyeao, 2020-05-08:
same here, works in Marlin mode, but not touch mode

stefi01, 2020-05-09:
i just installed the latest firmware and it now worked on both my lcds :) touch, non touch and usb

oldman4U, 2020-09-03:
Hi.

Please close this ticket which covers a solved issue.

Thank you

oldman4U, 2020-09-11:
@GenaUser 

Close this ticket

IIIxpHunterIII, 2020-10-06:
Dont close it, the Problem remains. A f*ing instruction manual would be nice tho.

oldman4U, 2020-10-06:
The question is if there is a problem with the firmware or something else. The user of the ticket does not respond and every other user is welcome to open a new ticket.

Regarding manuals. Feel free to provide them exactly the way you believe they should be available.

Thank you

closed 2020-10-06 by GenaUser

oldman4U, 2020-10-06:
@GenaUser I hope you have been able to solve the issue. Let me know if not.