bigtreetech/BIGTREETECH-TouchScreenFirmware #454: unknown command G29 at start and to long extrusion at end

JimGWIn, 2020-03-25:
### Description

### Steps to reproduce

1. [sense this update; at the start when selecting print the echo window pop up indicates "unknown command G29 "  ]
2. [at the end of the print the echo popup indicates "too long extrusion]

### Hardware Variant

### TFT Firmware Version 1.1.25.2.bin & Main Board Bigtreetech pro 1.1
[tftConfiguration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4382394/tftConfiguration.zip)
[G29.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4382487/G29.zip)

 Firmware marlin 2.0.5.1

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

bigtreetech, 2020-03-31:
it's because of "G29" in screen start gcode and "G1 E-4" in end gcode,  has improved it in #480

JimGWIn, 2020-03-31:
Thank you, I will look into it after dinner. Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/454#issuecomment-606333133",
"url": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/454#issuecomment-606333133",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

JimGWIn, 2020-03-31:
I disabled that feature and all is good. Re flashing the firmware also seems to have fixed my random reset problem. Thanks for the fast reply.

closed 2020-04-10 by bigtreetech