bigtreetech/BIGTREETECH-TouchScreenFirmware #455: [BUG] Disabling Start G-Code in Features causes all start code to be skipped?

Hukuma1, 2020-03-25:
### Description

I didn't want G29 at the start of every print, so I disabled the option in SETTINGS -> FEATURE menu.

```
/**
 * Start/End/Cancel G-code
 *
 * G-code that runs after a print starts, ends, or canceled.
 *
 * Disable in SETTINGS -> FEATURE menu.
 */
// Start G-code - run this G-code before starting print
#define PRINT_START_GCODE "G28\nG29\nG1 Z20\n" // Home all, run ABL routine, raise Z 20 mm.
```

However upon doing so it seems the printer is now ignoring my entire custom start Gcode from my slicer as well?

### Steps to reproduce

1. Disable TFT start G-Code
2. Try to print
3. Enjoy grinding sounds :/

**Expected behavior**
For the custom start GCode in the print file to execute.

**Actual behavior**
Code is skipped entirely? (also get "cold extrusion prevented" message)

### Hardware Variant
TFT35 with latest self-compile as of 03/25/2020

### Additional Information
Start code from Cura:

```
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
M420 S1 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
```

Hukuma1, 2020-03-25:
Seems it's doubling up on the start g-code between TFT firmware and slicer. Is this intended behavior?

thisiskeithb, 2020-03-26:
I’ve blanked out those custom start/end/cancel lines on my builds until it can be looked at more.

Start and end gcode is handled by your slicer, so I don’t know why it exists here.

Hukuma1, 2020-03-26:
Maybe it was intended to be as an override of some sort? In any case, perhaps it should be blank/commented out on default build? Easier to change the code in slicer than firmware anyway.

thisiskeithb, 2020-03-26:
This feature was added by @guruathwal, so they might have more background on why it was added.

guruathwal, 2020-03-26:
I checked the code and I can not find anything suspicious which can trigger such behavior.
Are you disabling the *Start g-code* and *End g-code* from the settings menu or Configuration.h?
Also, can you share the actual g-code from start to first few layers generated after slicing a model?

One thing to note is while the printer is performing the start g-code if you press the pause button it will send the retract command on pause and extrude command on resuming the print. this may also be the reason for the cold extrusion warning.

dineshvyas, 2020-03-26:
I think problem is in cura, I manually edit every file as follow.
First move before actual print start X, Y,  and Z move simultaneously.
G0 Xx Yy Zz Fffff
Edit that line as
G0 F2000 Xx, Yy
G0 F300 Zz.

thisiskeithb, 2020-03-26:
@guruathwal: What is the intended use case of start and end gcode in the TFT firmware? 

I’m not aware of a slicer that generates gcode *without* it, so it seems redundant.

If they will stick around, a better option would be to disable them by default so users don’t have doubled start/end scripts running every print.

guruathwal, 2020-03-26:
I can't remember who added the Start & End g-code and don't know why it was added. But I also had problems with it. 
Hence I added the options to disable them in the Feature-Settings menu without the need to recompile the firmware for the same. 
These settings have been working perfectly and no extra g-code is sent when they are disabled in settings.

thisiskeithb, 2020-03-26:
I’ll look into the code again and see if they can be disabled by default or remove them.

The cancel gcode makes sense to keep around since the TFT is acting as the host in that case.

Hukuma1, 2020-03-26:
> Are you disabling the _Start g-code_ and _End g-code_ from the settings menu or Configuration.h?

Disabling in Settings menu. If I only disable in Settings menu (and leave default code in Configuration.h), it bypasses ALL start G-Code and I cannot print anything. The only solution right now is a work-around to blank out the Start G-Code in TFT Configuration.h and leaving setting enabled in menu.

> Also, can you share the actual g-code from start to first few layers generated after slicing a model?

```
;FLAVOR:Marlin
;TIME:3996
;Filament used: 3.99473m
;Layer height: 0.2
;MINX:1.518
;MINY:99.001
;MINZ:0.2
;MAXX:233.484
;MAXY:136
;MAXZ:9.4
;Generated with Cura_SteamEngine 4.5.0
M82 ;absolute extrusion mode
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

; Ender 3 Custom Start G-code
M140 S60 ; Set Heat Bed temperature
M190 S60 ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
M420 S1 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M104 S200 ; Set Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
M109 S200 ; Wait for Extruder temperature
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:47
;LAYER:0
M107
G0 F6000 X17.035 Y99.259 Z0.2
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X17.746 Y99.157 E0.02389
G1 X18.207 Y99.106 E0.03932
G1 X18.967 Y99.048 E0.06467
G1 X19.749 Y99.016 E0.0907
G1 X20.071 Y99.008 E0.10141
```

> One thing to note is while the printer is performing the start g-code if you press the pause button it will send the retract command on pause and extrude command on resuming the print. this may also be the reason for the cold extrusion warning.

Never hit the pause button. I think it may be the issue with M109/M190 not working properly? It probably tries to execute command before going up in temperature.
`M109 S200 ; Wait for Extruder temperature`

Perhaps issue is linked to any waiting codes like https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/190#issuecomment-559604946 or new issue https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/458

thisiskeithb, 2020-03-26:
It was easy enough to change the default state of the start/end gcode in `Settings.c` from 1 to 0:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/523c3f54c6421352ea33406abd28a0e33d879384/TFT/src/User/Menu/Settings.c#L25-L26

See my PR with the change which disables those settings by default: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/461.

dp250f, 2020-04-01:
I'm seeing something similar with today's build. I also do not have Start or End code enabled (in TFT35) - I do that in Cura. It does process Cura's start code, but after a G29, it skips the next 2 or 3 lines.

bigtreetech, 2020-04-10:
@dp250f  fixed in #559.

dp250f, 2020-04-11:
> @dp250f fixed in #559.

Can confirm this is fixed. Thanks!

closed 2020-08-19 by bigtreetech