bigtreetech/BIGTREETECH-TouchScreenFirmware #457: [BUG] Problems in reading onboard SD (an SKR e3 Mini 1.2) with my TFT35-E3 v3.0 in touchscreen mode

Daelril, 2020-03-25:
I have the latest firmware installed, 25.2. Basically i can't get anything right if i chose to print from the onboard sd. The screen doesn't show all the gcodes, if i try to print something the temperature rises but it's not displayed, and if i try to stop the print it hangs on Loading... Also, i have a folder in that sd that i can't access and the name is not displayed, instead 
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4383193/Configuration.zip)

it shows only ???

Everything works fine if i put the sd in the screen slot, or if i use the Marlin emulation mode.

### Hardware Variant
TFT35-E3 v3.0
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.25.2, Marlin 2.0.5

[image]
[image]
[image]

gilliands, 2020-03-25:
This is an old bug reported some months ago.  There is a PR for a memory fix.  While it isn't perfect, it does improve file handling.  It comes down to inappropriate use of malloc() to claim temporary workspace and not freeing it in the order it was claimed.  This causes fragmentation of the RAM and then the heap runs over and then bang... things like your issue.
The best thing to do would be to stop using malloc except for initial claim and to reuse buffers instead of freeing and claiming all the time.

bigtreetech, 2020-03-26:
Hello, Maybe #416 can help you,  

> it shows only ???

about this problem, Is your file name made up of legal ASCII english characters? If there are some non ASCII characters, the encoding processed and fedback by Marlin is not the correct UTF-8, so it cannot be displayed normally

Daelril, 2020-03-26:
I will try that, thanks.
No special characters in the name of the folder. It's actually named "OLD" without quotiations

oldman4U, 2020-09-11:
@Daelril 

Any news regarding the reported issue?

Please let us know and close this ticket in case the problem is solved.

Thank you

closed 2020-10-06 by Daelril