bigtreetech/BIGTREETECH-TouchScreenFirmware #460: [Q] Why cannot upgrade firmware (BIGTREETECH TFT35 v1.2)

ionu7z1992, 2020-03-26:
Why cannot upgrade firmware on my tft35 v1.2??

BIGTREETECH TFT35 v1.2

I follow all the steps written in the readme(i coppied all the files required to the sd card),i plugged the sd card into the board but nothing happends(is starting normally in older version that already has in it)...
I mention that the sd card is fat32 formatted and is recognised by the board(i can print from it)

Hukuma1, 2020-03-26:
Can you show us a photo of what's on the card if you look at it from your computer?

ionu7z1992, 2020-03-26:
[image]

ionu7z1992, 2020-03-26:
i check with 2 sd cards+ 2 microsd(with sd adapter)..the same thing

Hukuma1, 2020-03-26:
Try this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/24#issuecomment-598692126

ionu7z1992, 2020-03-26:
It's working now...thank you

ionu7z1992, 2020-03-27:
One more question....looks like the eeprom does not store values...
Trying to set the ABL babystepping ...i set in up...tap on save a echo message : "settings stored (604 bytes; crc 23452) ok" show up on the screen but when i am reseting it(unplug) the value goes back to zero...
Do i miss something??
Thank you in advance

Clowntrigger, 2020-03-27:
What board do you have? It would regulate whether the values are stored in eeprom or not. You may need to see if there's a bug in the board firmware you're running.

Hukuma1, 2020-03-27:
Do the baby steps get saved to main z offset maybe? Can you check by first checking your z offset value and then when you save the new baby steps whether or not your z offset changes.

closed 2020-08-19 by bigtreetech