bigtreetech/BIGTREETECH-TouchScreenFirmware #463: [Q] Time for a new release?

thisiskeithb, 2020-03-26:
@bigtreetech The last official release was [Vx.x.25](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.25)/Nov 27, 2019, so we're probably due for another one with all the recent updates.

ssombra, 2020-03-27:
If I want to try all the recent updates, which firmware version should I use?

thisiskeithb, 2020-03-27:
> If I want to try all the recent updates, which firmware version should I use?

For the most up to date version, you’ll have to [compile your own](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md#firmware) firmware.

ssombra, 2020-03-27:
Sure, I can do that, my question was more about where are the most up to date files to compile. 
If I use the files here: 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
Am I using the most up to date files?

thisiskeithb, 2020-03-27:
> Am I using the most up to date files?

Yes.

ssombra, 2020-03-27:
Thanks, I'll give it a try.

kiwipaulrob, 2020-04-02:
I want to compile with the updates. Tried using Github desktop but it found lots of conflicts. Can someone with more knowledge than me explain how to resolve conflicts and generate a new bin file|?

thisiskeithb, 2020-04-02:
> I want to compile with the updates. Tried using Github desktop but it found lots of conflicts. Can someone with more knowledge than me explain how to resolve conflicts and generate a new bin file|?

Depending on how much you’ve customized your firmware, it’ll be easier to download the [master branch](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/master.zip) and transfer any settings over in `Configuration.h` using an app like Meld. Or just start from scratch since there isn’t much to configure to get up and running.

closed 2020-04-11 by thisiskeithb