bigtreetech/BIGTREETECH-TouchScreenFirmware #465: [FR] IDEA: Iconpack 'The Round Miracle' - Update for SW v25.3

Acenotass, 2020-03-26:
Hello everybody!
I created a fixed Iconpack in the release of a new version of the firmware in which the names of the image files changed.
Use only for firmware versions above 25.3
The Iconpack suitable for all versions of screens (2.4-2.8-3.5 inches) and compatible with Сlassic or Unified themes.
[image]
[TFT24 Iconpack 'The Round Miracle' by Acenotass for v25.3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4389142/TFT24.Iconpack.The.Round.Miracle.by.Acenotass.for.v25.3.zip)
[TFT28 Iconpack 'The Round Miracle' by Acenotass for v25.3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4389144/TFT28.Iconpack.The.Round.Miracle.by.Acenotass.for.v25.3.zip)
[TFT35 Iconpack 'The Round Miracle' by Acenotass for v25.3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4389145/TFT35.Iconpack.The.Round.Miracle.by.Acenotass.for.v25.3.zip)

dliving2010, 2020-04-12:
The Iconpack looks like a big improvement.  How do I replace my current Icons with these?  I am using a TFT35 V2 with the BIQU_TFT35_APP1_V2.0.26 update.

Thank you,
Dan

Acenotass, 2020-04-12:
@dliving2010 This is iconspack suitable for your firmware version

oldman4U, 2020-05-04:
Hello Acenotass.

I really like your theme and use it on a TFT35 and a TFT28. Unfortunately some images have been renamed with the last version of the TFT firmware and it seems that also the fix (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/641) will not make your theme work again. Could you please be so kind and check, so that everything works fine with the next firmware again.

Please let me know if there is anything i can help with.

Thank you

oldman4U, 2020-05-05:
Sorry to bother you again.

I just tested v26.1 with your theme and it works perfectly using TFT28 and TFT35.

Thank you

AntoszHUN, 2020-05-09:
Thank you very much! Very super! TFT35 V3.0

AntoszHUN, 2020-05-09:
What if we put it in the TFT display firmware? I just did it for the new firmware to be good. Because they renamed a few icons again.

cwizard, 2020-05-09:
> Hello everybody!
> I created a fixed Iconpack in the release of a new version of the firmware in which the names of the image files changed.
> Use only for firmware versions above 25.3
> The Iconpack [...]

This looks great.  Some new files have been added since v26.1 and a couple files have been renamed, so your theme pack needs a little bit of repair.  It looks outstanding!

kw3-research, 2020-06-26:
Nice work, just please note 3 icons are "missing", (the ones for extruder %)
On the zipped file, the name of the icons is now ok, and work well on a TFT24.

[bmp.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4836407/bmp.zip)

Cheers !

oldman4U, 2020-06-26:
Hi kw3-research.

Maybe you want to upload the new version which covers the latest release of the firmware from here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/681

oldman4U, 2020-10-09:
Hi. I believe you can remove the packages for all the old firmware versions. 

Thank you

closed 2020-10-09 by Acenotass