bigtreetech/BIGTREETECH-TouchScreenFirmware #468: [BUG] (EMERGENCY_PARSER kills the printer, hotend temperature)

juanyunis, 2020-03-27:
### Description
Printer gets killed by being unable to heat while enabling EMERGENCY_PARSER.

### Steps to reproduce

1. Installed the LCD
2. Boot printer and make sure LCD works
3. Put SDCARD with firmware update on the LCD
4. Flash new LCD firmware (custom with CLEAN_MODE_SWITCHING_SUPPORT)
5. There is a warning message saying that I needed to enable EMERGENCY_PARSER in Marlin
6. Flash printer firmware with EMERGENCY_PARSER enabled
7. Test heaters, nozzle and bed they do work and reach the temperature
8. Try printing from Octoprint
9. Reaching standby temperatures for nozzle and bed works
10. Homing, bed leveling works
11. Trying to reach nozzle extrude temperature fails
12. Printer gets killed and disconnected

**Expected behavior**
I expect the nozzle temperature reach the selected temperature for extrusion.

**Actual behavior**
Printer gets killed trying to reach selected temperature for extrusion. If i re-flash the printer firmware by disabling EMERGENCY_PARSER it does work, however the warning message will be stuck in the LCD.

### Hardware Variant
BIGTREETECH TFT35 E3 V3.0
### TFT Firmware Version & Main Board Firmware details
LCD firmware BIGTREE_TFT35_V3.0.25.3.bin
SKR 1.3 Firmware Marlin-bugfix-2.0.x

### Additional Information
Configuration.h https://pastebin.com/yQvjF75Y

guruathwal, 2020-03-27:
EMERGENCY_PARSER` is a Marlin feature which detects certain commands as they enter the receive buffer, so they cannot be blocked like **M108, M112, M410**.

Only the `EMERGENCY_PARSER` warning is displayed when the TFT detects that EMERGENCY_PARSER is disabled Marlin Firmware. Apart from this, there is no difference in the behavior of the TFT display if `EMERGENCY_PARSER` is disabled or enabled. 
With `EMERGENCY_PARSER` disabled in marlin firmware, these G-codes (M108, M112, M410) will not work as expected. Like the stop button on TFT sends Emergency-Stop G-code (M112) to halt the printer. Or some users use M0/M1 in gcode files to pause printing at a specific point, to resume printing through TFT the M108 command is used, which requires `EMERGENCY_PARSER` to be enabled to work.

This looks like a Marlin issue if heating is failing after enabling `EMERGENCY_PARSER` in marlin as no G-code command can induce heating failed error on marlin firmware. You also need to check Marlin firmware documentation for Thermal Protection here: [link](https://marlinfw.org/docs/configuration/configuration.html#safety) which state:
>For false thermal runaways not caused by a loose temperature sensor, try increasing WATCH_TEMP_PERIOD or decreasing WATCH_TEMP_INCREASE. Heating may be slowed in a cold environment, if a fan is blowing on the heat block, or if the heater has high resistance.

[image]

thisiskeithb, 2020-03-27:
Since `EMERGENCY_PARSER` isn’t available on `STM32` (ie: most of BTT’s boards), I’m wondering if there should be changes to that requirement.

Rogero-, 2020-03-30:
> Since `EMERGENCY_PARSER` isn’t available on `STM32` (ie: most of BTT’s boards), I’m wondering if there should be changes to that requirement.

last night i updated to latest code version BIGTREE_TFT35_V3.0.25.3.bin and i also had the warning message about the "Emergency Parser" being disabled in Marlin, i went to compile Marlin with Emergency Parser enabled and i discovered that it's not yet implemented for STM32, so i went back to TFT code and disabled the warning about the "Emergency Parser"

you need to edit the file: "..user\Menu\setting.c" and comment out 3 lines (55,56,57)
i will attach the edited file and a compiled Firmware for those who can't compile it themselves.

[image]

[BIGTREE_TFT35_V3.0.25.3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4402889/BIGTREE_TFT35_V3.0.25.3.zip)

Rogero-, 2020-03-30:
this should be disabled permanently in next public release or alternatively implement Emergency_Parser for STM32 in Marlin.

thisiskeithb, 2020-03-30:
Implementing emergency parser on `STM32` won’t be an easy task, so changing the code here to work around it will probably be a lot quicker.

@guruathwal / @Msq001 Do you have any ideas for what’s the best path moving forward?

Rogero-, 2020-03-30:
then commenting the 3 lines in settings.c to disable the warning is the way to go, it disabled the warning and all working fine for me.

thisiskeithb, 2020-03-30:
It looks like some TFT features rely on `EMERGENCY_PARSER` being enabled in Marlin, so those will need some extra code to account for hals like `STM32` that don’t support it.

guruathwal, 2020-03-30:
The warning was added to because of the features that rely on `EMERGENCY_PARSER`, specifically for M108 which I was going to add for canceling the print when the printer is at the heating loop (M190).
without which the heaters are left ON if the print is canceled while in the loop. while the user may think that all heaters are off. it does not disabled any feature on the TFT.

>this should be disabled permanently in next public release or alternatively implement Emergency_Parser for STM32 in Marlin.

@Rogero-  Marlin Firmware is a separate development managed by another team. you should open an issue in https://github.com/MarlinFirmware/Marlin/issues for this. 
Even when EMERGENCY_PARSER is not available on all boards it is still available for other boards. should they not use it?

> It looks like some TFT features rely on `EMERGENCY_PARSER` being enabled in Marlin, so those will need some extra code to account for hals like `STM32` that don’t support it.

@thisiskeithb There is no way of knowing which MCU the MainBoard has. So only way is to check if EMERGENCY_PARSER is enabled or not.
I will try to make this a one-time warning on the first run after the firmware update to make it less annoying.

PR #545 (2020-04-09): Fix Emergency_Parser warning to show only on first boot and settings …
Fix Emergency_Parser warning to show a Pop-up instead of showing in Infobox only on first boot and settings reset.

Resolves #538, resolves #468
closed 2020-04-09 by bigtreetech