bigtreetech/BIGTREETECH-TouchScreenFirmware #470: [FR] IDEA:  External Configuration.h File

Acenotass, 2020-03-28:
Communication on Facebook for many users is very difficult to independently assemble the firmware. I suggest the configuration.h configuration file be removed outside the firmware.

thisiskeithb, 2020-03-28:
How would you compile firmware if it was moved?

It’s easy enough to link/reference directly: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h

Acenotass, 2020-03-28:
Use all configuration values ​​from EEProm

thisiskeithb, 2020-03-28:
I see. Is this request different than https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/360?

Acenotass, 2020-03-28:
Sounds like the same idea.

thisiskeithb, 2020-03-28:
Since it's a duplicate request, mind closing & discussing in #360?

closed 2020-03-28 by Acenotass