bigtreetech/BIGTREETECH-TouchScreenFirmware #472: [BUG] (BTT TFT35 v3.0 - Status Screen)

adzhsv, 2020-03-28:
### Steps to reproduce

1. Select Print From On Board SD Card
2. Select File to print

**Expected behavior**

Heat Hotend, & Bed 

Hot End Temp, Set then show Temp increasing. 
Bed Temp, Set Then show Temp Increasing.
Fan Speed, Show Fan Speed

Once Temp reached, Set Temp same as actual temp Start Printing.

Printing Starts, Layer status increases as Z height increases

Finish Printing, Hot End & Bed Cool Down Status shows temp decreasing.

**Actual behavior**

Heat Hotend, & Bed 

Hot End Temp, Set Temp remains on 0 
Bed Temp, Set Temp remains on 0
Fan Speed, Fan Speed shows 0

Printing Starts, Layer status stays at 0 as Z height increases
After first Layer Fan set to 100%

Finish Printing, Hot End & Bed Cool Down Status shows temp same as set point, Never Changes.

### Hardware Variant
BTT TFT35 v3.0
BTT SKR MINI v3.0

### TFT Firmware Version & Main Board Firmware details
TFT Firmware - BIGTREE_TFT35_V3.0.25.2.bin
Marlin firmware - Marlin-2.0.x-SKR-Mini-E3-V1.2

### Additional Information

Using Stock Firmware in Repository for both screen and board.

Even when I compile Board firmware from Marlin Bugfix 2.0.x I have the same issue. Here is config file file when compiling Marlin Bugfix 2.0.x

[Configuration.h.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4396953/Configuration.h.zip)

guruathwal, 2020-03-28:
you need to update your TFT firmware to 25.3

valerox84, 2020-03-28:
The same happens to me, it gets blocked when you send it to print from the plate.
My configuration is:
bigtreetech e3 tft
v3
skr e3 dip v1.0
and use marlin 2.0.x bugfix

adzhsv, 2020-03-28:
> update

Hi guruathwal,  Where do I get 25.3,  Can only see 25.2

adzhsv, 2020-03-28:
Hi Valerox84,

Seems if you compile the firmware yourself the file you get is 25.3

There may have been some changes, that fix the issue, Ill report back once tested.

valerox84, 2020-03-28:
Hi
My problem is marlin, I am configuring 2.0.4, with that it does not give me an error when sending print from the plate, currently I had the 2.0.x bugfix, and with that it did not work for me.
Thanks for helping me a greeting

adzhsv, 2020-03-28:
Update,

Have updated TFT Firmware - BIGTREE_TFT35_V3.0.25.3.bin

Still have the same issue, When printing from Onboard SD (Motherboard SD) info screen not updated.

If i Print from TFT SD everything works as expected. Info Screen updates throughout print.

So problem is only seen when printing from On Board SD Card. (SKR Mini SD)

Thanks Adam

valerox84, 2020-03-28:
I corrected the problem by updating to a more recent marlin, for example in 2.0.1 it already works, but in the 2.0.x bugfix it doesn't work which was the one I had, and the board is similar to yours, which is a skr e3 dip.
I updated the screen with 25.3 and it still did not work and I put a preconfigured marlin that I made to a friend in 2.0.3 and if the screen works perfectly.
The problem comes from the Marlin.

Greetings

adzhsv, 2020-03-28:
Hi Valerox84,

Are you suggesting that I use 2.0.1 or 2.0.3 ?

Or should we go to latest release 2.0.5.2 ?

Appreciate your help :)

valerox84, 2020-03-28:
To me with 2.0.1 it is working correctly, the screen updated with 25.3, at least 2.0.1 I know that it is thinner than the others, but from 2.0.1 onwards the tft works correctly.
I suggest you do or marlin 2.0.1 through 2.0.3 on 4 and 5 will still have many bugs.
a greeting

adzhsv, 2020-03-29:
Hi Valerox84,

I think I have found the issue,

In Configuration_adv.h goto line 1143

  #if HAS_SDCARD_CONNECTION
    /**
     * Set this option to one of the following (or the board's defaults apply):
     *
     *           LCD - Use the SD drive in the external LCD controller.
     *       ONBOARD - Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
     *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
     *
     * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
     */
    #define SDCARD_CONNECTION LCD
  #endif

You can print from either SDCARD slot, However I believe that the firmware will only update on one, witchever is set, Not Both.

I'm now using the SDCard slot on TFT as it doesnt have char limit.

Thanks

closed 2020-08-19 by bigtreetech