bigtreetech/BIGTREETECH-TouchScreenFirmware #491: TFT35 V3.0 Bad assignment motherboard.

AntoszHUN, 2020-03-31:
Description of the problem.

The problem is with the TFT35 V3.0 display. As far as I know, this display has a STM32F207VC chip. However, the firmware uses the STM32F103VC chip translation. That's not good, I think. I'll tell you why.

- STM32F103VC: 72 MHz maximum frequency, up to 64 Kbytes of SRAM, 256 Kbytes of Flash memory

in contrast, the STM32F207VC is better.

- STM32F207VC: 120 MHz maximum frequency, Up to 128 + 4 Kbytes of SRAM, Up to 1 Mbytes of Flash memory

Is it possible to fix this? The display would probably work more reliably, accurately, and faster.

Look forward to your answers!

Sincerely!
AntoszHUN

[image]

[image]

thisiskeithb, 2020-03-31:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/355 Has more info on this.

AntoszHUN, 2020-03-31:
Ohh thank you!

closed 2020-03-31 by AntoszHUN