bigtreetech/BIGTREETECH-TouchScreenFirmware #494: [Q] Can M117 commands issued from Octoprint make it to the status screen in touchmode?

arminth, 2020-03-31:
I am using the TFT 2.4 on an SKR pro 1.1 connected to Octoprint on a RPI.
Any messages issued by M117 (e.g. percentage of print, layer number or anything else to be displayed via M117) are not making it to the the status screen in touchmode. XYZ coordinates, temperatures and fan percentages during printing on the other hand are displayed correctly.

What am I doing wrong!? Display is hooked up via TFT connector, EXP1 and EXP2.
Serial Port is configured to 6
Serial Port 2 is configured to 1

guruathwal, 2020-04-01:
You need to use M118

arminth, 2020-04-01:
That doesn't work either.

guruathwal, 2020-04-02:
If octoprint is connected directly to Mainboard then it will not work because it looks like marlin only relays M117 & M118 to the serial port it is received from.

oldman4U, 2020-10-09:
There are other users connecting octo directly using UART connection. Have you tried this?

oldman4U, 2020-10-22:
Hi.

There is a PR which seems to solve your reported issue. Do you want to give it a try: Fix to display M117 messages in info area #1189

Guten Abend

arminth, 2020-10-23:
I have tried the PR and it works!
Two things there:
It is mandatory to enable
´´´
#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
  #define HOST_START_MENU_ITEM  // Add a menu item that tells the host to start
#endif
´´´

in configuration_adv.h in Marlin!

I also found that the notifications received via M117 are not collected in the notification-window (if you tap on the notification, a list of all notifications pops up, which you can then clear. This list is empty even tough several M117 messages have been issued (and received).

Cheers
Armin

PR #1189 (2020-10-29): Fix to display M117 messages in info area
Fix to display M117 messages send by Octoprint plugin like DisplayLayerProgress in info area.

resolves #494
resolves #1050
closed 2020-10-29 by bigtreetech