bigtreetech/BIGTREETECH-TouchScreenFirmware #497: [BUG] Broken Fonts & Discolored Icons after #489

thisiskeithb, 2020-04-01:
### Description

#489 breaks fonts & icons are discolored on the TFT35 V3 & E3 (and possibly others, but this is all I can test with):
[image]

### Steps to reproduce

1. Update TFT35 V3 / E3 to 4e6d6f101ad5c9a34c4a6e8bcf01d18c99996408 or newer.
2. Observe broken fonts & discolored icons

**Expected behavior**
Non-broken fonts & icons

**Actual behavior**
Fonts are broken & icons are discolored.

### Hardware Variant
TFT35 V3 & E3

### TFT Firmware Version & Main Board Firmware details
25.3, self-compiled with default settings

Acenotass, 2020-04-01:
Also on my screen TFT3.5 v1.2

thisiskeithb, 2020-04-01:
Fixed in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/502.

closed 2020-04-01 by thisiskeithb