bigtreetech/BIGTREETECH-TouchScreenFirmware #507: [BUG] Z-Offset shows 0.00

Hans007a, 2020-04-01:
Hello,

in TFT Mode the Z-offset setting is shown a wrong value, it shows 0.00
In Marlin Mode the value is correct (I know Marlin mode has nothing to do with TFT Mode)

### Steps to reproduce

1. go into z-offset menu and look
2. do reset
3. go into z-offset menu and look again

**Expected behavior**
shown the right z-offset value

**Actual behavior**
shows 0,00

### Hardware Variant
TFT35 v3.0 with the last firmware, BTT SKR Pro v1.1 with the last Marlin bugfix

Regards

Hans

kingontheroad, 2020-04-01:
I had that yesterday, tested them all yesterday. And the fw of last March 12 still works best and also remember;)
fw 25.2 in the firmware archive folder

Hans007a, 2020-04-01:
I don't use the compiled images

kingontheroad, 2020-04-01:
the 25.2 that i mention is original and not compiled

bigtreetech, 2020-04-02:
hello, have fixed in #516

closed 2020-04-05 by Hans007a