bigtreetech/BIGTREETECH-TouchScreenFirmware #508: when entering new values in the Z offset the full value is not displayed.  If the value is 7.3, the displayed value is .3[BUG] (short description)

6988isgr8, 2020-04-01:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Interplain, 2020-04-05:
I have also found this to be the case. An it also doesn't let you put input the minus sign before the number. You have to put a number in first and delete the number and the minus will enter.

oldman4U, 2020-09-03:
This should work using a recent firmware

Please close this ticket

closed 2020-09-03 by 6988isgr8