bigtreetech/BIGTREETECH-TouchScreenFirmware #509: SKR 1.3 vs PC vs TFT module - communication/serials/USB confusion[Q]

bogus105, 2020-04-01:
Buys!
My setup: 
Marlin 2.0 bugfix based firmware + Bigtreetech SKR 1.3 controller board (LPC1768 processor) + 2.4" TFT controller moduler from Bigreetech (TVT24-V1.1). 
The facts:
SKR 1.3 controller board has: 
1. USB connector to connect with PC for printing from host (PC), upgrading firmware (sending firmware.bin from PC to SKR over USB cable, then reboot SKR), using as SD card reader (when connected with PC, SKR acts like a SD card reader so we can see what we have on the SD card inserted into the SKR's onboard slot).
2. SD card slot - as in 1 we can use it for firmware flashing, as SD card reader or to store print files we want to print in stand-alone mode

TFT24-V1.1 controller has:
3. SD card slot (big version) for flashing the firmware and storing print files and print from it (?) in standalone mode (?)
4. USB port for USB sticks - for storing print files and print from it (?)  

In Vision Studio PlatformIO we can see in Configuration.h and Configuration_adv.h files some settings for SDSUPPORT for serial communication etc...

The questions i have:
1. in configuration.h there is #define SERIAL_PORT -1 and #define SERIAL_PORT_2 0. What these ports really are? What particular ports are responsible for? What the available arguments ([-1, 0, 1, 2, 3, 4, 5, 6, 7]) mean exactly?

2. BAUDRATE - i heard i have to set it to 115200. Is it really the only option? During the print (so far only using PC as host over USB cable!) i can see very short (like miliseconds) stops resulting in small blobs all over my print. It looks like USB data transfer problem (Win10 USB problem?) - maybe the USB serial communication buffer shortage? 
3. If i change BAUDRATE, where do i have to change the setting as well? In my host software? I'm using CURA 4.5.  Is this setting (BAUDRATE) responsible for PC (host) - SKR communication ONLY?? 

4. How about communication between SKR and TFT moduler? I can see there is TX0/RX0 assigned for it on the SKR circuit board. Does it mean TFT module communicates with SKR via SKR's hardware serial 0? If yes is it the same serial as defined in Configuration.h file as `#define SERIAL_PORT`? Should it be -1, 0, 1 or what?

Ovearall design questions:
Correct me if i'm wrong. I uderstand it all might work like this: SKR can talk to onboard SD slot via hardware SPI (MISO, MOSI, SCK and CS pins). So it is a really FAST connection, actually the fastest one in tthe whole setup. The best one to use as main communication mean for PRINTING. So i can see it as tt is the best option to use SKR onboard SD slot for print files in terms of both reliability and speed. 
The next way to print something is to use TFT24-V1.1 module which has both SD card slot and USB port for USB sticks. HOWEVER as TFT module communicates with SKR via UARTs RX0/TX0 lines only it has inherent buffer and speed limitations. I do not know the exact fdesign but i assume that during the printing TFT module continouslly is sending G-codes from SD card/USB stick connected to TFT slots to SKR, and SKR is doing all the timing and pulses creation work. If womething happen between TFT module and SKR during the print - the print is ruined. In my opinion it is not the best option to use TFT module slots for print files storage and ANY of the comm problems between the two systems (one is SKR and the second one is TFT processor and its firmware) lead to print failure. 
The last way for rinting i'm assuming now is printing from host - Win10 based PC in my case over the USB cable. The same drawbacks as in previous way - weak comm stability and speed in general.

So i'm chosing the following approach:
1. slice the file in Cura1
2. send the g-code file from Cura directly to SD card inserted into the SKR montherboard slot using USB cable connection between PC and SKR (SKR acts like a regular SD card reader).
3. go to TFT touch screen mode, pick the file stored on the SD card on the SKR board and hit "PRINT".

The problem now is when i go to touchsceen and go for PRINT ->  OnboardSD i can see "Loading..." for a while and then a message: "Read onboard SD card error". 
When i go to: MENU -> Custom -> Init SD card i get echo reply: "SD init fail   ok".
But i do not know WHICH SD card we are talking about here? The one in TFT module or in the SKR?

I'm trying to figure out how to set everything up so i can easily use SKR onboard SD slot and controll my printer over USB as well.

Please lets start the discussion here and clarify the problems related to SD card, USBs and communication issue between SKR, TFT modules and hosts.

ssombra, 2020-04-01:
I am not sure, but I think that in order to print directly from the board SD card, you should specify in Marlin that you are using the onboard SD card instead of the LCD controller SD card:
#define SDCARD_CONNECTION ONBOARD
maybe I am wrong, I always print through Octoprint, but that's my understanding.

bigtreetech, 2020-04-02:
hello，about your question:
 1. #define SERIAL_PORT -1,  -1 means use USB virtual serial port，0 means hardware UART0, 1 means hardware UART1, etc... 
    so the define depend on the board design, SKR V1.3 USB wiring to LPC1768 usb pins, and TFT connector wiring to LPC1768 UART0 pins, so the SERIAL_PORT need be set to -1(USB) or 0(TFT connector)
    2. BAUDRATE doesn't have to be set to 115200, 115200 is just a common value, If the baud rate is too low, the data interaction will be slow. If the baud rate is too high, it is easy to be interfered and generate garbled code, resulting in printing failure
    3. if you changed BAUDRATE in Marlin,  You also need to change other BAUDRATE to be the same as Marlin, including all devices using SERIAL_PORT or SERIAL_PORT_2 to communicate with Marlin, such as: PC software(Cura, S3D, printrun, etc...), Octoprint, TouchScreen(TFT24: you can modify baudrate in "Settings" menu), etc...
    4. `#define SERIAL_PORT` value only depend on the designed of the motherboard run with Marlin(SKR V1.3)， The principle of TFT control printing is actually the same as that of PC, which is realized by serial port sending gcode commands and parsing feedback

bigtreetech, 2020-04-02:
In fact, M21/M20 is sent to get the gcode file through Marlin when you click TFT "Print->OnBoardSD", and the control of the SD card slot will be taken over by TFT if TFT24 work in touch screen mode，So Marlin can't access SD card at this time, and will feedback "SD init failed",  If you want to print from onBoardSD in this state, you need to `#define SDCARD_CONNECTION ONBOARD `in marlin, as the above friend said.

cooldudie2, 2020-06-15:
sorry to highjack this thread, but i need help with something that is similar. I want to connect a malyan screen to the skr 1.3 board using the tft header. What is the designated serial port of the the tft header and what name is used to describe the serial port. i need to know its designation as it  is required in marlin to use the screen. marlin calls for the port so i need to know the value to write. for example  is it called serial1, serial0, serial or MYSERIAL1 Myserial2, etc i cant get it to work at all. any help is appreciated here.

closed 2020-08-19 by bigtreetech