bigtreetech/BIGTREETECH-TouchScreenFirmware #512: Illuminated button no longer works after recompiling with new files?

BelSD, 2020-04-01:
Good evening,
When I compile with the latest files, I no longer have the option to light the command button. in the Option menu there is no longer the possibility to choose this color and actually turn it on.
Is this normal?

thisiskeithb, 2020-04-02:
Are you using `default_envs = BIGTREE_TFT35_E3_V3_0` in [`platformio.ini`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/platformio.ini)?

BelSD, 2020-04-02:
Ok I was wrong, thank you!
But now I have these 2 messages during compilation ...
[image]
[image]

thisiskeithb, 2020-04-02:
Those are just warnings and can be ignored. As long as you see the "SUCCESS" message, your build completed fine:

```prolog
==================== [SUCCESS] Took 2.42 seconds ====================

Environment            Status    Duration
---------------------  --------  ------------
BIGTREE_TFT35_V1_0     IGNORED
BIGTREE_TFT35_V1_1     IGNORED
BIGTREE_TFT35_V1_2     IGNORED
BIGTREE_TFT35_V2_0     IGNORED
BIGTREE_TFT35_V3_0     IGNORED
BIGTREE_TFT35_E3_V3_0  SUCCESS   00:00:02.416
BIGTREE_TFT28_V1_0     IGNORED
BIGTREE_TFT28_V3_0     IGNORED
BIGTREE_TFT24_V1_1     IGNORED
MKS_32_V1_4            IGNORED
MKS_32_V1_4_NOBL       IGNORED
==================== 1 succeeded in 00:00:02.416 ====================

Terminal will be reused by tasks, press any key to close it.
```

BelSD, 2020-04-02:
Thanks for your help, it works now ...

closed 2020-04-02 by BelSD