bigtreetech/BIGTREETECH-TouchScreenFirmware #515: [Q] Add Chamber Temperature at Unified Menu

tpipits, 2020-04-02:
I would like to know how can i modify the code so the chamber temperature can appear to main/print screen at unified menu and at the top right corner at the status bar.

I have trigoriila v1.1 board (based on ramps 14) with marlin 2.0.5.2 firmware and i have added a chamber temperature sensor. At marlin simulation screen the chamber temp appear fine and i want to add it and to touch mode. 

I have no problem to be combined with other icon (for example the bed temp icon to change between bed temp and chamber temp.

I am waiting for your help.
Thank you

tpipits, 2020-04-06:
After 4 days that have been past i couldn't find any solution on how to display chamber temp and fan to 2.5.3 firmware for tft 24v1.1 screen.

@Developers could you change the issue's label to enhancement and add this function as feature update?

Some printers have a heated or not heated chamber with a fan and a temperature thermistor and marlin can support these functions out of the box (at marlin simulator screen champer's temp and fan appear and function very well but when i use touch mode i can't see the chambers temp and i can't see status and make use of the fan).

Thank you.

bigtreetech, 2020-07-28:
chamber feature is merged in #875， so close here, if you have some else issue, please reopen here and contact me

closed 2020-07-28 by bigtreetech