bigtreetech/BIGTREETECH-TouchScreenFirmware #517: [BUG] MKS TFT 32 V4. 0 - unexpected interrupt when connecting a usb flash drive.

Uz45, 2020-04-02:
### Description

### Steps to reproduce

1. Insert USB flash drive

**Expected behavior**

**Actual behavior**

The device freezes.

Program
 received signal SIGINT, Interrupt.
WWDG_IRQHandler () at TFT\src\Libraries\Startup\stm32f10x_hd/startup_stm32f10x_hd.s:113
113	  b  Infinite_Loop
### Hardware Variant

default_envs = MKS_32_V1_4
### TFT Firmware Version & Main Board Firmware details

SOFTWARE_VERSION=25.3
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

rebel1, 2020-04-04:
Easy to fix
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/532

Uz45, 2020-04-04:
> Easy to fix
> #532

Great! It works.

closed 2020-04-09 by bigtreetech