bigtreetech/BIGTREETECH-TouchScreenFirmware #518: Avoiding the M105 command.

AntoszHUN, 2020-04-02:
Very annoying commands disappearing from the command line due to the automatic sending of the continuous M105 command. Thus, the queried info is not visible because it cannot be rolled back. I was wondering if the M105 command could be removed from the display firmware? The Marlin program already uses a command that can be linked to the M155 command. This sends the temperature data at set intervals. Couldn't this be used? Has anyone outside me ever thought of that?

Hi, AntoszHUN

guruathwal, 2020-04-02:
did you not see the Feature Settings Menu? There is an option to disable terminal ack.

AntoszHUN, 2020-04-02:
So what does ACK mean? What is ACK? Because I don't think many people know anything about it. Neither do I. Sorry for the idiot question! :)

hfog, 2020-04-02:
ACK is short for Acknowledgement. In this case it refers to a message sent over the serial bus to acknowledge receipt of an instruction. It's part of the communications protocol.

AntoszHUN, 2020-04-02:
Thank you very much for your reply! I understand now. : D

closed 2020-04-14 by AntoszHUN