bigtreetech/BIGTREETECH-TouchScreenFirmware #521: [BUG] compiler error 'undeclared' (detailed inside)

Hans007a, 2020-04-02:
Hallo,
here my compiler error

### Description

TFT\src\User\Menu\StatusScreen.c:52:101: error: 'SSICON_NAME_Y0' undeclared here (not in a function); did you mean 'ICON_PAGE_UP'?
   {1*ICON_WIDTH+0*SPACE_X+START_X - BYTE_WIDTH/2,    ICON_START_Y + 0 * ICON_HEIGHT + 0 * SPACE_Y + SSICON_NAME_Y0},
                                                                                                     ^~~~~~~~~~~~~~
                                                                                                     ICON_PAGE_UP
TFT\src\User\Menu\StatusScreen.c:58:93: error: 'SSICON_VAL_Y0' undeclared here (not in a function); did you mean 'SSICON_NAME_Y0'?
   {START_X + 0 * ICON_WIDTH + 0 * SPACE_X,  ICON_START_Y +  0 * ICON_HEIGHT + 0 * SPACE_Y + SSICON_VAL_Y0,
                                                                                             ^~~~~~~~~~~~~
                                                                                             SSICON_NAME_Y0
TFT\src\User\Menu\StatusScreen.c:75:121: error: 'STATUS_MSG_BODY_YOFFSET' undeclared here (not in a function)
Compiling .pio\build\BIGTREE_TFT35_V3_0\src\src\User\os_timer.o
 const  GUI_RECT msgRect ={START_X + 1 * ICON_WIDTH + 1 * SPACE_X + 2,   ICON_START_Y +  1 * ICON_HEIGHT + 1 * SPACE_Y + STATUS_MSG_BODY_YOFFSET,
                                                                                                                         ^~~~~~~~~~~~~~~~~~~~~~~
TFT\src\User\Menu\StatusScreen.c:76:121: error: 'STATUS_MSG_BODY_BOTTOM' undeclared here (not in a function); did you mean 'STATUS_MSG_BODY_YOFFSET'?
...

### Steps to reproduce

1. compile the new commit

**Expected behavior**
it compile like yesterday

**Actual behavior**
compiling error, wrong include?

### Hardware Variant
TFT35 v3.0

Regards

Hans

thisiskeithb, 2020-04-02:
I just pulled the latest code (db70304) from 11 hours ago and do not get this error.

Try cleaning your PIO install (click the trash icon) & compile again:
[image]

Hans007a, 2020-04-03:
I put:

`#include "variants.h"`

into the StatusScreen.h and now it works...

closed 2020-04-03 by Hans007a