bigtreetech/BIGTREETECH-TouchScreenFirmware #524: Vertical Screen GUI

ChengQian1007, 2020-04-03:
It will be super, if a vertical GUI is supported.

drphil3d, 2020-04-19:
Please Please Please Please

bigtreetech, 2020-08-19:
How's this?
[image]

[image]

OVDEN13, 2020-08-24:
how to get it?

ChengQian1007, 2020-08-26:
Cool! That is what I want!

ChengQian1007, 2020-08-26:
Thanks a lot!

hazeyjay77, 2020-08-28:
At last!!! I need i need i need please give me lol :)

9zdot, 2020-09-03:
Nice, that's exactly what I was looking for. 
When this feature will be available?
Thanks

9zdot, 2020-10-01:
Hey @bigtreetech do you have plans to release it? I'll be glad if you can give us more information what was changed so me and others can compile ourselves to use in our personal projects.

drphil3d, 2020-10-01:
@bigtreetech I love you man / woman

oldman4U, 2020-10-20:
Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-21 by ChengQian1007

SlackHacky, 2020-11-08:
Any update on release of portrait mode

oldman4U, 2020-11-08:
Hi. As far as I know there is currently no development on this. Are you a developer which could help to develop this?

radry, 2021-03-14:
Why is this issue closed? It hasn't been implemented yet.
@bigtreetech please release vertical GUI

oldman4U, 2021-03-14:
Hi radry.

Many abandoned tickets made this system unusable in the past. Therefore a bot has been installed which closes tickets which have not been used for more than 2 months. Unfortunately because of this, also important tickets like this one have been closed, so there were two new tickets introduced, where verified bugs and feature requests are collected. You can find the two tickets at the top of the front page of this repository.

Unfortunately I can not tell you when a vertical interface will be released, but probably the lack of open source developers who are willing to spend their spare time developing it will be a part of the reason. Are you a developer who is willing to help or is there anything else you believe you could help us with?

Hope this answers your question, please let me know

Thank you

radry, 2021-03-15:
@oldman4U 
That's not how git works. If you close an issue, people who want to fix them can't find or reference them correctly. The "collection" ticket is nice as an overview but can't replace the git issue system. @bigtreetech showed a picture of vertical interface, so obviously it exist. Why did they not release it on the repository?

oldman4U, 2021-03-15:
Feel free to make it better.

technoo10201, 2021-03-28:
Hi guys ! :)
Any news about this feature ? I'm really interested !
Here is a great magnetic support which i would like to use https://www.thingiverse.com/thing:4809101

Please let us know, we really want it @bigtreetech 

Regards
Technoo'

E-Studios, 2021-04-19:
Shameless topic bump, it's been over 1 year already... hows going with the vertical mode?

oldman4U, 2021-04-20:
How it goes depends on the availability of Users like you and me who are willing to spend their spare time to this open source software project. Are you a developer who can help?

kisslorand, 2021-04-20:
> Shameless topic bump, it's been over 1 year already... hows going with the vertical mode?

Tough words... I do not remember anyone ever promising anything about vertical mode.

E-Studios, 2021-04-20:
> > Shameless topic bump, it's been over 1 year already... hows going with the vertical mode?
> 
> Tough words... I do not remember anyone ever promising anything about the vertical mode.

The only thing you have to do is scroll to the top of this thread... And see the images BTT posted.
They even said in a mail conversation they are "working" on it. And is also on their TD list.

So these words aren't that tough...

> How it goes depends on the availability of Users like you and me who are willing to spend their spare time to this open source > software project. Are you a developer who can help?

No I'am not a developer who can help for this screen. I also wonder if it's  "Open Source" I see most of them just using a piece of software you can just design the GUI with dragging images and create buttons and such. So I wonder how BTT is really desigining these screens. taking the hard way or easy way? the screens are basically the same as MakerBase's screens.

oldman4U, 2021-04-20:
Maybe you are an interface designer or you have other skills which could help!? Maybe you can write manuals?

You can see that you can download the source code from this repository free of charge and guys like kisslorand and digant73 are improving this firmware free of charge. I call this open source maybe you have another word for it.

E-Studios, 2021-04-20:
> Maybe you are an interface designer or you have other skills which could help!? Maybe you can write manuals?
>
> You can see that you can download the source code from this repository free of charge [...]

Yes i'am a designer and mechanical engineer, but a newb "programmer". So the only thing I can do is make nice icons, images etc. So thats why I asked how the project was going.

I'am just asking BigTreeTech how it is going with their Vertical/Portrait Mode progress, not to stir things up...

There is a request from their customers to get a Vertical/Portrait Mode, BTT showed pictures of their progress last year, added it to their todo list.

And for "Open Source"  I mean. lot of the Chinese manufacturers say its "OpenSource" but its not really that opensource. So that's why i wonder if this screen's software/firmware is really opensource.

oldman4U, 2021-04-20:
But it seems that you are our man - maybe;-)

The problem is the following. It is not about rotating the screen 90%, it is doing so not for a single dedicated piece of hardware, but for different screens. So it should work for:

TFT24/28 which has a screen resolution of 320x240 and therefore an aspect ratio of 4:3
TFT35 which has a screen resolution of 480x320 and therefore an aspect ratio of 3:2
TFT43/50 which has a screen resolution of 480x270 and therefore an aspect ratio of 16:9
TFT70 which has a screen resolution of 800x480 and therefore an aspect ratio of 10:6

And there is not a single layout but screens with buttons, buttons and bars, lines with text, lines with text and a checkbox or scroll buttons. Additional text which is shown sometimes and overlay windows and so on and so on. You know the different screens.

Not to forget the limited resources, hardware and man power wise;-)

I am willing to help, but alone this will probably take another year or so until a usable design is defined. Maybe I am wrong, but I believe that a real nice vertical user interface is several years away, or needs more users working on it. And I am not sure that there is a good way to implement this on all of the screens available from BTT. Which brings me back to the beginning of this comment. Maybe it would be good to implement this only for a single screen....

radry, 2021-04-20:
@oldman4U 
As you can see above bigtreetech has already developed a vertical screen layout but they don't share it yet. This is their project, they designed the hardware, software and sell it to make profit, so they should release it and not we as users develop it ourselves.

kisslorand, 2021-04-20:
Guys, it is not as easy as it may seem. It might take up too much resources from the TFT's limited storage capacity so maybe a separate project is needed. This matter has already been raised in the case of having everything in one single FW. There has been discussions also about having RRP, Klipper, etc on a separate project. The demands are endless and the space is limited.
The more complicated is a project it is more susceptible to errors, bugs, more harder to implement a new feature for every scenario/variant/version, more harder to find and eliminate bugs.
I am more and more strongly against having everything in one single FW.

oldman4U, 2021-04-21:
@radry What you see above are two screens on a single TFT, where the two screens are very similar in what they show. Unfortunately there are approximately 20 different screens layouts which all have to be adjusted to work on a vertical screen and this is like kisslorand wrote above, a lot of work. And he has to know it, because he is one of the most active developers here. I guess we are talking about at least one year full time job and for those still not understand, believe or haven't heard it - the hardware comes from BTT and also the foundation of the firmware, but all the nice interfaces and features you see are from those users which are mentioned here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/graphs/contributors.

Every user has certain skills which could be helpful to improve the solution. To sit and wait and to ask for something does not help. Let's see if E-Studios is willing to spend a part of his valuable spare time.

radry, 2021-11-10:
I started digging into this issue.  
Rotating the screen to portrait mode is easy but it comes with two problems which I still couldn't solve:

1) The touch layer is not rotated, only the LCD screen orientation.
I haven't found where in the code the touch layer/hardware orientation is specified. 

2) The layout of the menu is hardcoded and isn't well documented. Meaning it tries to display 8 icons in two rows (4 each row). This results in an error (display goes into listening mode) when the specified horizontal resolution isn't enough to fit the row.
(When only rotating the screen but not adjusting the resolution it will display the menu cut off.)

The menu is hardcoded in the file TFT/src/User/API/menu.c and menu.h
It needs trial and error to make progress.

Sim098, 2023-05-30:
i"am using TFT70 touch screen with Octopus pro motherboard. i want to remove some of the icons from the display. for e.g. onboard_sd, TFT_sd, cz_fade etc  I'm not using these icons anymore so want to delete them from the screen. Did anyone know how to do it ??

kisslorand, 2023-06-05:
For the moment there's no other way but changing the source code and compile it yourself.