bigtreetech/BIGTREETECH-TouchScreenFirmware #528: TFT24 no printer attached

rosariovw, 2020-04-03:
Someone know how to fix this? 

When installed, first i didn't get a thing out of the screen. After some reading here, i updated the marlin, and the firmware of the TFT screen. After this, i got it working for 3 or 4 hours, and then lost connection. Since then, i don't get it to connect. Also, noticed after that my setting aren't saved on the board, not my pre-heat settings or mu E-steps, after every reset, i have to reconfigure.

oldman4U, 2020-09-03:
Hi.

What is the status of this ticket?

Please let us know or close it if not needed anymore

rosariovw, 2020-09-03:
Hi, got it solved, changed to the tft50, that solved it

met vriendelijke groet

oldman4U, 2020-09-03:
Cool. Great TFT

Please close this ticket then

Thank you

closed 2020-09-03 by rosariovw