bigtreetech/BIGTREETECH-TouchScreenFirmware #531: [BUG] Current code breaks knob LED

Taomyn, 2020-04-04:
Latest source update has hardcoded LED values for the knob (sets to 1) that seem to override anything I put into the user config. I use 6 for GREEN, but I see warnings in the compile that the value is being overwritten and then when placed on my TFT35 the knob is not lit and I cannot see any menu setting to change it. I reverted back to a build from 30/03/2020 and it's working again.

guruathwal, 2020-04-05:
which firmware version did you used?

Taomyn, 2020-04-05:
The current code that you can download - last commit was:
[image]

thisiskeithb, 2020-04-05:
Make sure you're `default_evns` to `BIGTREE_TFT35_E3_V3_0` and not `BIGTREE_TFT35_V3_0` in `platformio.ini`:

```ini
[platformio]
src_dir      = TFT
boards_dir   = buildroot/boards
default_envs = BIGTREE_TFT35_E3_V3_0
```

thisiskeithb, 2020-04-05:
Looks like theres a bug in `SanityCheck.h`'s logic for the LED color when using the incorrect environment/TFTs that don't support it.

guruathwal, 2020-04-05:
@thisiskeithb  but it does not seems like it. I set the value to 6 and the logic works like it should:

@Taomyn you also need to reset the settings to see settings set through `Configuration.h file`

[image]

thisiskeithb, 2020-04-05:
@guruathwal: Try setting `default_envs` to `BIGTREE_TFT35_V3_0` and compile again.

guruathwal, 2020-04-05:
@thisiskeithb  that's strange here is my output:

[Link](https://pastebin.com/YW82nasE)

thisiskeithb, 2020-04-05:
You're compiling with `BIGTREE_TFT35_E3_V3_0`. Compile with the (incorrect) `BIGTREE_TFT35_V3_0` environment.

`SanityCheck.h` should ignore those checks completely when compiling for non ILI9488 LCD types now that we have `LCD_DRIVER_IS()` macros.

guruathwal, 2020-04-05:
@thisiskeithb  some defines are requied to set the default setting for `InfoSettings` even if those settings are not read by some variants. 

The same Define can also be seen here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/db70304e640c02f20bf4bf30f6895bfe9084bd74/TFT/src/User/main.c#L49-L53

thisiskeithb, 2020-04-05:
Undefining `STARTUP_KNOB_LED_COLOR` before setting it to `1` for the non-E3 LCDs takes care of the warning, but there's probably a more elegant way to do it:

```cpp
#ifdef LED_color_PIN
  #ifdef STARTUP_KNOB_LED_COLOR
    #if STARTUP_KNOB_LED_COLOR < 1
      #error "STARTUP_knob_LED_COLOR cannot be less than 1"
    #endif
    #if STARTUP_KNOB_LED_COLOR > 9
      #error "STARTUP_knob_LED_COLOR cannot be greater than 9"
    #endif
  #else
    #undef STARTUP_KNOB_LED_COLOR
    #define STARTUP_KNOB_LED_COLOR 1
  #endif
  #else
    #undef STARTUP_KNOB_LED_COLOR
    #define STARTUP_KNOB_LED_COLOR 1
#endif
```

Taomyn, 2020-04-06:
Hi, so it seems my change to environment variable got missed when I reset my downloaded code - not sure how it happened. When I did that everything was fine with my knob LED again. Thanks @thisiskeithb 

However it does seem I inadvertently did find an issue with the LED code for the other environments, so not a wasted bug report in the end.

closed 2020-04-16 by Taomyn

DeepVoid, 2020-05-30:
> Make sure you're setting `default_envs` to `BIGTREE_TFT35_E3_V3_0` and not `BIGTREE_TFT35_V3_0` in `platformio.ini`

It works!!! :-)