bigtreetech/BIGTREETECH-TouchScreenFirmware #533: [BUG] TFT35 v3.0 Flash disk error if there is more than 1 file

samgozman, 2020-04-04:
### Description

For two days I racked my brain - why was I able to print 1 cube, but I couldn't print anything more? I tried everything - moving jumpers, different firmware, tried all the flash drives in the house, etc. The problem turned out to be much simpler - TFT35 can't start printing from a flash drive if it has more than one file on it. LoL

### Steps to reproduce

1. Create multiple gcode files on usb
2. Insert USB in TFT35
3. Try to start printing

**Expected behavior**
Printing started

**Actual behavior**
"Read U Disk error!" or it just freezes

### Hardware Variant
TFT 35 v3.0 plus SKR Pro v1.1
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.25.3, Marlin 2.0.1

### Additional Information
* Stock firmware

samgozman, 2020-04-07:
The next day, the display completely stopped reading anything from the usb.
[Video demonstration](https://youtu.be/o5uvRDo-g5Y)

Hukuma1, 2020-04-08:
I have this phenomenon on my TFT35 with mini E3.

The workaround for me has been (with Cura Slicer):

Once you eject the SD card and it does not print when you click on the filename, put it back into your computer, delete the gcode file, and copy it to card again (save to removable). Then eject it once more and put it into TFT. It should print as normal then.

PR #544 (2020-04-09): comment out preview model function for print from U disk
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

fixed #533 
This is just a temporary solution. it can print normally from u disk in my testing. More debugging is needed to find the root reason
### Benefits

### Related Issues
closed 2020-04-09 by bigtreetech