bigtreetech/BIGTREETECH-TouchScreenFirmware #534: TFT35 V3 Esp8266 WiFi does not work properly

Neme77, 2020-04-05:
### Description

<The wifi module in the TFT35 V3 does not work correctly, it stops and the commands arrive with considerable delay making it necessary to restart the machine
I have loaded the index file correctly, I regularly access the printer but it crashes with every action. -->

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

bigtreetech, 2020-04-11:
Hello, it should be fixed in #565, you can download the lastest version firmware and test it

Neme77, 2020-04-11:
Great, now all working fine....
Big Thx...

closed 2020-04-11 by Neme77