bigtreetech/BIGTREETECH-TouchScreenFirmware #535: [BUG] TFT24 keep trying to install any file in the sd card

edoardoo, 2020-04-05:
## Description  
After firmware is installed, when the reset button is pressed the display will try to flash any of the files in the sdcard. No matter the name or the extension, the display will try to install it, resulting in the message "Illegal FLASH APP!" and a bricked device.

### Steps to reproduce

1. Flash a firmware by placing it in the root folder of the sdcard
2. Restart with the sdcard still inside the display
3. 'Illegal FLASH APP!' message appears

**Expected behavior**
The display boots normally after restart

**Actual behavior**
The device is bricked and a new firmware need to be installed

### Hardware Variant
TFT24 v1.1
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT24_V1.1.25.3 with Marlin 2.0.X

### Additional Information

guruathwal, 2020-04-05:
The bootloader on the MCU of the TFT will search for any file with the text '.bin' anywhere in the name. 
make sure no file with '.bin' in the name exists

edoardoo, 2020-04-05:
Sorry but "no matter the file extension" means that if I put an empty "foo.txt" file inside the sd the TFT will try to install it, failing.

To be more clear, after flashing a firmware,if I leave the sdcard inside and power cycle, the display will try to install the backup files, failing.

guruathwal, 2020-04-06:
ok.  did you tried Re-formatting the SD card to Fat32 and copying only the firmware file, not even the TFT24 folder?

edoardoo, 2020-04-07:
Yes, I tried that too. 
After installing only the firmware (without the TFT folder), I can confirm that the behavior remains the same.

bigtreetech, 2020-04-09:
Hello, What is the capacity of the SD card you use? Can you provide a video of TFT24 updating any SD card file to flash? This is a very abnormal behavior, because the bootloader of TFT24 only recognizes the files of named "BIGTREE_TFT24 xxxxx .bin" to update.

edoardoo, 2020-04-09:
Hi, 
I tried again today, after upgrading the BTT mini E3 v1.2  with Marlin 2.0.5.3 and, incredibly enough, I can now use the SD and the USB. 
I've tried to load two random gcodes on the very same sd card and it doesn't seems to have problems anymore. The USB stick is working too.
I've also tried both in Marlin mode and Btt mode (switching cables). Everything works as expected in both modes, except some random white screen when using the touch screen.
I am 100% sure that it wasn't working before the Marlin upgrade.
I'd even tried compiling the display firmware by myself, with the same sad outcome.
Now, the only difference is that I compiled the Marlin 2.0.5.3 board firmware with the `btt_512k_usb` variable in platformio.ini.  
Please note that the display firmware hasn't changed since before the flash!

Just to sum it up: 

- Display keeps trying to flash everything is on the card (even reset.txt)
- Flash the display latest standard firmware
- Switched to display Marlin mode (with gray cable)
- Upgrade board Marlin to 2.0.5.3
- Sd card and USB stick works
- Switched to BTT mode (with black cable)
- Sd card and USB stick works

I'm available if you need any other info to address this.

closed 2020-04-09 by edoardoo