bigtreetech/BIGTREETECH-TouchScreenFirmware #536: [BUG] Unable to print from USB (and from sdcard)

edoardoo, 2020-04-05:
### Description

I'm unable to print from a USB stick.  Whenever I try a read disk error appears.
No matter the usb drive (I've tried three different models of three different sizes). 
On about 40 tries, I only managed to get it to start printing once.
Given the fact that I can't use the sdcard because the device thinks that every file on it is worth trying to be installed, and consequently fails bricking itself (see #535), this really leaves me with **no options to print a file from this display**.

### Steps to reproduce

1. Take a USB stick with a GCODE file on it 
2. Insert it into the display
3. Try to start a print

**Expected behavior**
The print process is being initiated

**Actual behavior**
Sometimes the USB disk is not being read at all, while some other times when the files list is loaded and a file is being tapped the display blinks and nothing happens. 

### Hardware Variant
TFT24 v1.1
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT24_V1.1.25.3 Marlin 2.0.x

### Additional Information
I've used a mac to load the files and deleted anything different than the single file and `.Trash` folder

samgozman, 2020-04-05:
Hi! I have a similar problem #533 
Try to format the USB flash drive and write only 1 file to it.  Also make sure the SD card slot is empty.  Only in such a combination I could make this unfortunate piece of tech to work

edoardoo, 2020-04-05:
Hi!
Last night before writing the isssue I've read #533 and tried immediately withoyt any luck.  
This morning I tried again following this process:
- format the card to FAT
- remove all the automatically created folders except the `.Trash` which can't be removed
- add a small calicat.gcode to the usb
- insert the usb stick inside the display
- try to print the calicat gcode

Results (yeah, it's not deterministic): 
- the screen goes white (a reset is needed)
- even worse, the display starts (without any confirmation) printing, immediatly sending the gcode commands to extrude without waiting for the printer to get sufficiently hot. While doing this it constantly proposes an alert about "Cold extrusion prevented" (ironic)

samgozman, 2020-04-05:
I'am using [SD Formatter](https://www.sdcard.org/downloads/formatter/) in macOS for all USB flash and SD cards. If I do not confuse anything, then it also deletes the `.Trash` folder.
I could only print if:

- There are only 1 file
- Select a file in the first 30 sec after initializing the printer. Otherwise the screen goes black until reset.

Sanguium, 2020-04-05:
Maybe same problem as https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/444

edoardoo, 2020-04-07:
I confirm that no matter what, I can't print from a standard sd card on this display.

samgozman, 2020-04-07:
It seems my luck has come to an end.  Today, the printer does not read any usb flash drives, sd cards neither in the display nor in the skr pro 1.1 board.
Also, USB control never worked on skr pro itself, so I don’t even have the ability to print via octoprint.
In other words, thanks to the products of the @bigtreetech, I cannot start printing on the printer in any of the existing ways.  I don’t know how to restrain anger!

edoardoo, 2020-04-08:
I managed to solve the mini e3 usb connection problem by merging the latest
Marlin release 2.0.5.3 and compiling using the btt_512K_USB default_envs.
Hope it helps.

closed 2020-04-09 by edoardoo