bigtreetech/BIGTREETECH-TouchScreenFirmware #537: [BUG] Unable to tap on some menu elements when printing

edoardoo, 2020-04-05:
### Description

While printing the printer temps and fan speed are represented correctly but tapping on them does not produce any result. While some of these elements are reachable by other ways,  some others are not, like adjusting the flow rate (for what I've seen).

### Steps to reproduce

1. Start a printing a file (in my case from the controller board sd card file)
2. Try to change the flow rate from the home view

**Expected behavior**
The flow rate page (or any other of the quick links) opens

**Actual behavior**
Nothing happens

### Hardware Variant
TFT24 V1.1
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT24_V1.1.25.3 Marlin 2.0.x

### Additional Information
Due to #535 #536 printing from the controller board sd card is the ONLY option I have.

guruathwal, 2020-04-05:
That's because they are not buttons.. you need to go to the 'More' menu to adjust the setting while printing.

edoardoo, 2020-04-05:
Ok, I missed that `percentage` meant speed & flow settings.
Thanks

closed 2020-04-05 by edoardoo