bigtreetech/BIGTREETECH-TouchScreenFirmware #540: [BUG] TFT not reading filmament detector when connected to SKR

planecrazy29, 2020-04-06:
With the Smart filament detector connected to the SKR it will work in Marlin mode but not touch screen mode.
With the 12v UPS connected to the SKR it will work in Marlin mode but not touch screen mode.  Is there a way to get the screen to read these pin states?

### Steps to reproduce

1. Print something
2. Run out of filament
3. Works in Marlin mode but not in Touch Screen mode

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

bigtreetech, 2020-04-10:
hello,
 * you need connect Smart Filament Sendor to TFT "FIL_DET" port not SKR board if you print from TFT SD in touch mode, according to here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/312#issuecomment-609520483
* 12v UPS: you need only insert in SKR board and it will raise z axis when power failed,  TFT will always save the printed data in real time

closed 2020-11-03 by stale[bot]

oldman4U, 2020-12-21:
Please try the following:

Send M75 to Marlin when print begins. Marlin doesn't know that the print is started so no filament runout warnings. It's not a bug.
You can put M75 in your Start-Gcode and you're set. That's what I did.