bigtreetech/BIGTREETECH-TouchScreenFirmware #541: [Q] RGB <> BGR on TFT24

dchauran, 2020-04-07:
Hello - I am changing to a different screen on my tft24 board, and everything works, except that red and blue are inverted. Looking at the code and the data sheet for ILI9341, it seems like I need to change the value of register 0x36, however no matter what I put in for the data value for 0x36 nothing changes.

I'm looking at line 274 of LCD_init.c and the relevant section from the datasheet is attached.

Any idea what I should change to switch between RGB and BGR?

[image]

bigtreetech, 2020-04-07:
Hello, try to change the value of register 0x36 in here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/cff1164003c26deabbf1d747a2ec354a21e3b671/TFT/src/User/Hal/LCD_Init.c#L531
here will refresh the value of registe  0x36 that has setted in `void LCD_init_RGB(void)`

dchauran, 2020-04-07:
Thank you! I missed that one. It's working perfectly now.

closed 2020-04-07 by dchauran