bigtreetech/BIGTREETECH-TouchScreenFirmware #542: TFT35 V2 - Printing From the Card Reader Causes Zits (Also known as Blobs & Pimples)

dliving2010, 2020-04-07:
### Description of the Issue

The TFT35 V2 causes Zits (Also known as Blobs & Pimples) on a radius when printing from the card reader.  On every print, even a simple single walled cylinder, when printing from the card reader built into the TFT35 V2 (current firmware), the hotend will pause at random times (at least every 10 seconds) for less than 1/2 second.  This is enough time to cause an extra amount of filament to be extruded to create what looks like a zit on the radius of a print.  This never happens when printing a straight line in either axis.

### Work Around

To get around this issue, though not the prefered or convenient choice, is to print from the card reader built into the control board (in my case, the SKR V1.3).  When printing from the internal card reader of the control board, the hotend never pauses on the radius.  While there are still some minor zits, these are most likely due to a layer change.  As an additional benefit, I also noticed faster print times, which is most likely due to the time savings that came from the reduction in the hotend pausing every few seconds. 

**Expected behavior**

In order to prevent artifacts like zits, pimples & blobs, the Hotend should move smoothly when printing.

**Actual behavior**

The hotend pauses every few seconds when printing a radius (not when printing a straight line).

### Hardware Variant

SKR V 1.3 (Marlin 2.0 (bugfix-2.0.x))
TFT35 V2.0 (TFT35_V2.0.25.1)

thisiskeithb, 2020-04-07:
Update the TFT to the latest firmware (25.3) and the latest bugfix-2.0.x from today.

Early TFT code was incorrectly polling Marlin too often for X/Y/Z position and it’d cause pauses when printing.

dliving2010, 2020-04-12:
> Update the TFT to the latest firmware (25.3) and the latest bugfix-2.0.x from today.
> 
> Early TFT code was incorrectly polling Marlin too often for X/Y/Z position and it’d cause pauses when printing.

Thank you.  That was exactly the issue.  Before I updated, I went ahead and printed from the internal card and got much better results; although there were still zits on the print.

However, after updating the TFT35 firmware (to the now current version of V2.0.26), I am happy to report that the are ZERO zits on the print!

closed 2020-04-12 by dliving2010

Sebazzz, 2020-09-30:
@thisiskeithb I'm running into this issue with my BIQU B1 as well - and I flashed [firmware 2020-08](https://github.com/bigtreetech/BIQU-B1/tree/master/Firmware/2020-08-18) for the touch screen - but I also suspect this touch screen is slightly different so I can't just flash any new firmware from this repository I assume. 

Do you happen to know if this change was also released in the 2020-08 touch screen firmware for BIQU B1?